Zepmanbc/creopyson #1: Exception thrown in `creo_list_dirs`

timoguic, 2019-07-03:
* Creopyson version: 0.3.1
* Python version: 3.6
* Operating System: Windows

### Description

I tried to list the directories in the current session with `creo_list_dirs()` but I received an exception telling that Creoson did not return the right data.

### What I Did

```
>>> c.creo_list_dirs()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Tim\.virtualenvs\creopyson_test-yJfFsqNW\lib\site-packages\creopyson\creo.py", line 111, in list_dirs
    return client._creoson_post("creo", "list_dirs", data, "dirlist")
  File "C:\Users\Tim\.virtualenvs\creopyson_test-yJfFsqNW\lib\site-packages\creopyson\connection.py", line 81, in _creoson_post
    raise MissingKey("Missing `data` in creoson return")
creopyson.exceptions.MissingKey: Missing `data` in creoson return
```

Zepmanbc, 2019-07-03:
Hi @timoguic, thanks for your issue!
I see the problem, creoson does not return `data` if there is no folder in your current directory.
I will fix it asap

timoguic, 2019-07-03:
Hmm, yes - the directory was empty. I created an empty folder and It is working now. :)

PR #2 (2019-07-03): Issue #1 fix
creo_list_dirs() now return an empty list if there is no folder in the cirectory

Close #1
closed 2019-07-03 by Zepmanbc