Zepmanbc/creopyson #10: creopyson.is_creo_running() throws exception if creo has not been started

saturley, 2020-02-11:
Creopyson 0.4.1
Python 3.8.1
Microsoft Windows 10.0.16299.1625
Creoson 2.5.0
Creo 4.0 M100

### Description

creopyson.is_creo_running() throws exception if creo has not been started.

### What I Did

1. Start Creoson server without Creo running
2. Issue the following commands:
    creo_session = creopyson.Client()
    creo_session.connect()
    creo_session.is_creo_running() -> Throws an exception: RuntimeError: Unable to connect to Creo through J-Link
3. Start Creo, issue creo_session.is_creo_running().  This returns True (as expected).
4. Quit Creo, issue creo_session.is_creo_running().  This returns False (as expected).

Zepmanbc, 2020-02-11:
Hi @saturley , thank for your issue

Creopyson returns an exception because Creoson return 'error' : True.

Here are the differents Creoson returns:

Creo "never" been run
creoson returns : `{'status':{'error': True, 'message': 'Unable to connect to Creo through J-Link'}}`

Creo is running
creoson returns : `{'data': {'running': True}, 'status': {'error': False}}`

Creo has been closed
creoson returns : `{'data': {'running': False}, 'status': {'error': False}}`

As you can see with the first case there is an error because the connection had never been made.
I think this is a conception choice from Creoson, maybe @davidhbigelow or @adama2000 could tell you more about this with theses informations.

adama2000, 2020-02-11:
I'll investigate the creoson side.

adama2000, 2020-02-11:
I've checked in a change to creoson that fixes the issue.  We haven't built a new release yet though.

In the meantime, one workaround I discovered was if you issue your connection.is_creo_running *before* you do a connection.connect, it won't throw an exception like that.

Zepmanbc, 2020-02-12:
thanks @adama2000  🙏

Zepmanbc, 2020-02-13:
@adama2000 will it returns `{'data': {'running': False}, 'status': {'error': False}}` ?
Because without `data` it will still raise an exception in creopyson because if `error` is False there must be `data`.
That's what i've analysed from the creoson doc when i started creopyson.

adama2000, 2020-02-13:
It should still return that, yes.

Zepmanbc, 2020-03-08:
correction in creoson [v2.6.0](https://github.com/SimplifiedLogic/creoson/releases/tag/v2.6.0)

closed 2020-03-08 by Zepmanbc