Zepmanbc/creopyson #11: Creo freezes after file_open when using start_creo to launch Creo up.

terryxychan, 2020-02-14:
* Creopyson version: 0.4.1
* Python version: 3.7.4
* Operating System: Windows 10

### Description

I'm trying to open a file from Windchill using file_open() and Creo froze. 

### What I Did

I tried 3 methods:
#### Method 1
The first method is to launch up Creo with the nitro_proe_remote.bat
```
c = creopyson.Client(ip_adress="localhost",port=9056)
c.connect()
c.start_creo("path_to_nitro_proe_remote.bat")
```
I made sure Creo is properly launched up before running these lines:
```
c.windchill_authorize("username", "password")
c.file_open(file_="something_cool.asm")
```
And Creo frozes. 
The nitro_proe_remote.bat looks like this:
```
@echo off
"c:\work\PTC\Creo_3.0\M120\Parametric\bin\parametric.exe" "c:\work\PTC\Creo_3.0\M120\Parametric\bin\parametric.psf" %*
```
#### Method 2
The second method is to launch up Creo with a different nitro_proe_remote.bat:
```
@echo off
"c:\work\PTC\Creo_3.0\M120\Parametric\bin\parametric.exe" %*
```
Ran the same thing as Method 1 and the same happens.
#### Method 3
The third method is to launch up Creo directly using parametric.exe:
```
c = creopyson.Client(ip_adress="localhost",port=9056)
c.connect()
os.system("C:\\Work\\ptc\\Creo_3.0\\M120\\Parametric\\bin\\parametric.exe")
```
And this time, I was able to open the file without Creo freezing. 

Does anyone have any idea why this is happening?

Zepmanbc, 2020-02-16:
Hello @terryxychan 
1. Do you have the same issue with method 1 and 2 after the success of method 3? I'm wondering about Windchill's cache.

2. Do you have issue with method 1 and 2 if you don't connect to Windchill and open a local file?

3. With method 1 or 2, can you list workspaces or files? (just to verify connection is ok and the problem is located on opening file)

terryxychan, 2020-02-17:
> 1. Do you have the same issue with method 1 and 2 after the success of method 3? I'm wondering about Windchill's cache.

Yes, I tried again and the same thing still happens. 

> 2. Do you have issue with method 1 and 2 if you don't connect to Windchill and open a local file?

Yes, I'm having the same issue with 1 and 2 while trying to open a local file without connecting to Windchill.

> 3. With method 1 or 2, can you list workspaces or files? (just to verify connection is ok and the problem is located on opening file)

Yes, I'm able to list my workspaces and files in Windchill after authorizing.

Zepmanbc, 2020-02-26:
I cannot reproduce your problem... I can open Creo and files with Creoson.

Do you launch Creoson with *CreosonSetup.exe* or *creoson_run.bat* ?
I've noticed that the *creoson_run.bat* console window's name is changing with my *nitro_proe_remote.bat* batch name.
I did not take care and closed the console, then creson was closed, maybe you did the same?

terryxychan, 2020-02-26:
@Zepmanbc,

I launched creoson using creoson_run.bat and I did not close the terminal. I'm not sure what you meant by "creoson_run.bat console window's name is changing with my nitro_proe_remote.bat batch name." I haven't noticed any change in my creoson_run.bat console window name. 

I've also tried launching creoson using CreosonSetup.exe, the same thing happens.

davidhbigelow, 2020-02-26:
What version of Creo?

Does the Playground work in CREOSON's help web page?

terryxychan, 2020-02-26:
@davidhbigelow 

If I launch up Creo through exe and authorize to Windchill server manually, I was able to open the file through playground. But not if I launched through nitro_proe_remote.bat and authorize to Windchill server manually, it freezes.

davidhbigelow, 2020-02-26:
Ok - that is good to know that it works via the Playground.

Are you doing any windchill functions with CREOSON - like

windchill : set_server
windchill : authorize

prior to trying to open any file?

terryxychan, 2020-02-26:
@davidhbigelow 

Yes, I'm doing windchill: authorize before opening a file.

davidhbigelow, 2020-02-26:
Ok - forgot to ask....

Are you also setting a WORKSPACE before attempting the open?

windchill : set_workspace

terryxychan, 2020-02-26:
@davidhbigelow 

Currently no, but I would like to add it in the future

davidhbigelow, 2020-02-26:
Well ... logically - your last workspace should be active on your
connection (you would think)...

But you may have to set it before running the open command.

Try that ... if it doesn't work... send us an email support@nitromation.com
and we can get on an on-line meeting to take a closer look.

davidhbigelow, 2020-02-26:
OH - also.... talking it over with the developer.... are you pointing to
the same startup batch /executable file that you run manually.

It could be an environment difference if that is not the same thing - or
starting in the same location as you normally do (e.g. config.pro not being
used when started up via the .bat file you are trying to run with...)

Just a hunch.

terryxychan, 2020-02-26:
I'm still experiencing this, I'll email you through that email. 
The startup batch is indeed pointing to the correct executable file.

adama2000, 2020-02-26:
If it doesn't work when you start Creo via nitro_proe_remote.bat, but does when you start it manually, is there a difference between the two?

The nitro_proe_remote.bat should be calling whatever is called when you start Creo manually -- whether it's a .exe, a .bat file, or something else.  And with the same command-line arguments.

One thing we have done elsewhere is to try running the nitro_proe_remote.bat manually from the command line, and see if the problem still occurs; if it does, tweak nitro_proe_remote.bat until it goes away.

davidhbigelow, 2020-02-26:
If you want to take a look at it now... we can do an on-line meeting if you
are up for it.

terryxychan, 2020-02-26:
@adama2000 

Hmm, I started Creo by manually running nitro_proe_remote.bat and it worked....

terryxychan, 2020-02-26:
@davidhbigelow 

Sure, how do you want to meet?

terryxychan, 2020-02-26:
@davidhbigelow 

Actually, do you mind having this meeting at 2/27 at 1pm EST?

davidhbigelow, 2020-02-26:
go to creoson.com ... request a meeting....

adama2000, 2020-02-28:
Terry, when you do the c.file_open what options (such as display=True) are you passing to the function ?

terryxychan, 2020-02-28:
@adama2000 
Hey Adam,
I'm using the default values which I believe is:
c.file_open(file_, dirname=None, generic=None, display=None, activate=None, new_window=None, regen_force=None)

Zepmanbc, 2020-05-22:
This issue is still unresolved? @terryxychan

terryxychan, 2020-06-03:
@Zepmanbc It wasn't resolved last time I saw it, but it works really well with Nomad when CREO is opened directly through the command line.

Zepmanbc, 2020-06-29:
I close but if anyone got the same needs and problems i will reopen

closed 2020-06-29 by Zepmanbc