Zepmanbc/creopyson #14: Unsuccessful connection to Creo from Creoson when launching Creoson up with python

terryxychan, 2020-03-24:
* Creopyson version: 0.5.0
* Python version: Python 3.7.4
* Operating System: Windows 10

### Description

I'm launching Creoson with a python script using Popen. This script is located at the Creoson folder. 

The reason for doing this is to integrate it with Nomad for managing Creoson. Nomad stops services by sending out SIGINT signals. 
```
#!/usr/bin/env python3
import sys, os
import subprocess
import signal
import time
import psutil

java_command = ["jre/bin/java","-classpath",\
    ".;commons-codec-1.10.jar;.creo4funcs-1.7.0.jar;creofuncs-1.7.0.jar;creoson-2.9.0-javadoc.jar;creoson-core-2.9.0.jar;creoson-intf-1.9.4.jar;creoson-json-1.6.0.jar;creoson-jsonconst-1.6.0.jar;CreosonServer-2.6.0.jar;jackson-annotations-2.0.5.jar;jackson-core-2.0.5.jar;jackson-databind-2.0.5.jar;C:/Work/ptc/Creo_3.0/M120/Common_Files/text/java/pfcasync.jar",\
    "-Dsli.jlink.timeout=200",\
    "-Dsli.socket.port=9056",\
    "com.simplifiedlogic.nitro.jshell.MainServer"]

def sigterm_handler(signal, frame):
    global p
    # graceful shutdown
    # save the state here or do whatever you want
    print("terminate processes")
    try:
        p.terminate()
    except:
        print("Creoson not launched")

    process = psutil.Process(p.pid)    
    PROCNAME = process.name()
    i = 0
    while i <= 5:
        for proc in psutil.process_iter():
            if proc.name() in PROCNAME:
                print(proc.name())
                proc.kill()
        i += 1
    sys.exit(0)

signal.signal(signal.SIGBREAK, sigterm_handler)
signal.signal(signal.SIGINT, sigterm_handler)

def main():
    global p
    print("Opening Creoson")
    DETACHED_PROCESS = 0x00000008
    CREATE_NEW_PROCESS_GROUP = 0x00000200
    p = subprocess.Popen(java_command,creationflags=DETACHED_PROCESS | CREATE_NEW_PROCESS_GROUP)
    # stdout, stderr = p.communicate()
    print(p.pid)
    while True:
        time.sleep(10)

main()
```
The environmental variables required by Creoson is all set to System. 
I am able to start and stop Creoson and I am able to access the webpage through `localhost:9056`.

### What I Did
However, when I start Creoson this way, I'm running into this problem:
```
>> c = creopyson.Client(ip_adress="localhost",port="9056")
>> c.connect()
>> c.file_get_active()
Traceback (most recent call last):
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1336, in getresponse
    response.begin()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\util\retry.py", line 400, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1336, in getresponse
    response.begin()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 49, in _creoson_post
    r = requests.post(self.server, data=json.dumps(request))
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\file.py", line 269, in get_active
    return client._creoson_post("file", "get_active")
  File "C:\Users\YChan14\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 51, in _creoson_post
    raise ConnectionError(e)
ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))
```
And Creoson exited after this. 

I believe this is due to the connection lost between creopyson and Creoson? 

I wonder if there is a better way to start Creoson through python. 

Thanks!

Zepmanbc, 2020-05-22:
Hi @terryxychan 
I add logging on request/responses in release 0.5.1 maybe it can help for your problems

terryxychan, 2020-06-03:
Thanks, @Zepmanbc! Unfortunately, I'm off the project and won't be working at it anymore. But I will relay this information to my former manager.

Zepmanbc, 2020-06-29:
I close but if anyone got the same needs and problems i will reopen

closed 2020-06-29 by Zepmanbc

Zepmanbc, 2020-10-07:
Hi anyone who try to run Creoson from Python
Here is a solution that works for me:

```
import os

os.chdir(r"C:/path/to/creoson/folder/")
os.startfile(r"creoson_run.bat")
```

You need to have run _CreosonSetup.exe_ one time to setup configuration, then you will be able to run Creoson with _creoson_run.bat_ in the same folder.

DieSwartKat, 2020-10-08:
We are doing something similar to this. I should have mentioned it before.
Thanks for posting it.

We use it with a django server which we start up with CREO also from a
batch file. We found that if CREO gets restarted the original Django server
is still running. So we also include this script to kill the server, to
make sure we only have one server running at a time.

FOR /F "tokens=5 delims= " %%P IN ('netstat -a -n -o ^| findstr :9056') DO
wmic.exe process where ProcessID=%%P delete

We are using port 9056 for the Django server.