Zepmanbc/creopyson #16: file_save() not saving when list provided

Osiel-Gaona, 2020-06-26:
* Creopyson version: 0.5.1
* Creoson version: 2.6.0
* Python version: 3.7.2
* Operating System: windows 10
* Creo 6

### Description

I was trying to save a list of files, the files where previously copied and renamed and all are in creo memory 
when using file save for the list it did nothing, when i do it in for loop it does save them.

### What I Did

*This method did not work for me
a = creo.file_list()
print(a)
['t5_pin.asm', 't5_boss.prt', .... etc (i removed the rest of files to save space)']
creo.file_save(a)

*this way it does work
a = creo.file_list()
for i in a:
creo.file_save(i)

Zepmanbc, 2020-06-29:
Hi @Osiel-Gaona 

Indeed save with a list of files does not works.

Here are the tests i've made:

I use *part1.prt* and *part2.prt*

## Test 1 : all 2 parts opened and modified (part2 active)

run request
```
{
  'sessionId':'-5989841723074638635',
  'command':'file',
  'function':'save',
  'data':{'files': ['part1.prt', 'part2.prt']}
}
```
clean up session
open files => **only part 2 saved** 👎 

## Test 2 : all 2 parts modified and closed (still in session)
run request (the same as above)
clean up session
open files => **nothing saved** 👎 

## Test 3 : only 1 part(just for check)
modify part1
close file (still in session)
run request
```{
'sessionId':'-3335764160595805866'
'command':'file'
'function':'save'
'data':{'file': 'part1.prt'}
}```
clean up session
open file => **part1 saved** 👍 

@davidhbigelow @adama2000 do you have an idea?

adama2000, 2020-06-29:
Looks like it's a bug in creoson.  I'll post a fix for it.

Adam Andrews

adama2000, 2020-06-29:
A bug fix is now posted to creoson.

Zepmanbc, 2020-06-30:
Thank you for your reactivity! 👍

closed 2020-07-01 by Osiel-Gaona