Zepmanbc/creopyson #17: file_get_transform() is not working

31friedl, 2020-07-07:
* Creopyson version: 5.1
* Creoson version: 2.6
* Python version: 3.8
* Operating System: Win 10

### Description
Try to get the transformation of the main coordinate system.

Code:
import creopyson

c = creopyson.Client()
c.connect()
c.is_creo_running()

f=creopyson.feature_user_select_csys(c)

print (f)
print (creopyson.file_get_transform(c, csys=f['featlist'][0]['name']))

### What I Did

{'featlist': [{'name': 'ASM_CS0', 'type': 'csys', 'feat_id': 58}]}
Traceback (most recent call last):
  File "H:/python/automatic-model-generation/creo-export/creopyson_test.py", line 12, in <module>
    print (creopyson.file_get_transform(c, csys=f['featlist'][0]['name']))
  File "C:\Users\friedlw\AppData\Roaming\Python\Python38\site-packages\creopyson\file.py", line 372, in get_transform
    return client._creoson_post("file", "get_transform", data, "transform")
  File "C:\Users\friedlw\AppData\Roaming\Python\Python38\site-packages\creopyson\connection.py", line 84, in _creoson_post
    raise MissingKey(
creopyson.exceptions.MissingKey: Missing `transform` in creoson result

Zepmanbc, 2020-07-07:
Hi @31friedl 

I've tried without args and a random file:

Request:
```
{
    'sessionId': '-5409027441229007638', 
    'command': 'file', 
    'function': 'get_transform', 
    'data': {}
}
```

Response:
```
{
    'status': {'error': False}, 
    'data': {
        'y_rot': -0.0, 
        'origin': {'x': 0.0, 'z': 0.0, 'y': 0.0}, 
        'z_axis': {'x': 0.0, 'z': 0.0, 'y': 1.0}, 
        'y_axis': {'x': 0.0, 'z': -1.0, 'y': 0.0}, 
        'x_axis': {'x': 1.0, 'z': 0.0, 'y': 0.0},
        'x_rot': -90.0, 
        'z_rot': -0.0
    }
}
```
Indeed `transform` is not in the result as expected in the exemple: https://github.com/SimplifiedLogic/creoson/blob/master/creoson-server/web/assets/creoson_stuff/jsonSpecs/file-get_transform.json

@adama2000 @davidhbigelow can you tell if `transform` is expected ?

if not, I will update creopyson to remove this key

adama2000, 2020-07-07:
You're right that the documentation doesn't match the return value.  That's my fault.

I could either change the function's return or change the documentation; I figure it's safer to change the documentation in case anyone else is already using that function.

adama2000, 2020-07-07:
If you update creopyson to remove the transform key, you'll also update it to return the 7 new keys instead right?

Zepmanbc, 2020-07-08:
No, it will return the « data » part

```
>>> import creopyson
>>> c = creopyson.Client()
>>> c.connect()
>>> c.file_get_transform()
{'y_rot': -0.0, 'origin': {'x': 0.0, 'z': 0.0, 'y': 0.0}, 'z_axis': {'x': 0.0, 'z': 0.0, 'y': 1.0}, 'y_axis': {'x': 0.0, 'z': -1.0, 'y': 0.0}, 'x_axis': {'x': 1.0, 'z': 0.0, 'y': 0.0}, 'x_rot': -90.0, 'z_rot': -0.0}
```
I'll do the update ASAP

Zepmanbc, 2020-07-08:
@31friedl you can write your code like this:

```
import creopyson

c = creopyson.Client()
c.connect()
c.is_creo_running()

f=creopyson.feature_user_select_csys(c)  ==> f=c.feature_user_select_csys()

print (f)
print (creopyson.file_get_transform(c, csys=f['featlist'][0]['name']))  ==> print (c.file_get_transform(csys=f['featlist'][0]['name']))
```
I know the documentation shows the client as an argument, it is because of the structure, if i had to do it again now i would change this to be more clear.
And with a builder pattern it would be cleaner than all these imports: https://github.com/Zepmanbc/creopyson/blob/master/creopyson/__init__.py

31friedl, 2020-07-08:
Thank very you!

{'y_rot': -0.0, 'origin': {'x': 0.0, 'z': 0.0, 'y': 0.0}, 'z_axis': {'x': 0.0, 'z': 0.0, 'y': 1.0}, 'y_axis': {'x': 0.0, 'z': -1.0, 'y': 0.0}, 'x_axis': {'x': 1.0, 'z': 0.0, 'y': 0.0}, 'x_rot': -90.0, 'z_rot': -0.0}

I'll do the update ASAP

adama2000, 2020-07-08:
I've fixed the doc/examples for the creoson function.

closed 2020-07-09 by Zepmanbc

31friedl, 2020-07-10:
Hallo Zepmanbc,

thank you for your effort, I will try it.

Best