Zepmanbc/creopyson #19: Template name passed to drawing in WindChill with drawing_create

victoman1234, 2020-07-17:
* Creopyson version: 0.5.2
* Creoson version: 2.6.0
* Python version: 3.7
* Operating System: Windows 10 Pro 64-bit

### Description

Using the drawing_create command to create and save to windchill a new drawing with the Part Name as the Drawing Name.

The drawing is created properly but the name of the selected Drawing Template is used as the name of the new drawing in WindChill. The new drawing has the correct File Name, has the correct Number but does not have the correct name in WindChill.

###What I did

```
import creopyson
c = creopyson.Client()
c.connect()
c.server
cprint = c.sessionId
print("\n" + "session Id = " + cprint)
c.drawing_create("bmh_d_eng_detail")
c.disconnect()
```

[image]

davidhbigelow, 2020-07-17:
> [image: 2020-07-17 14_31_02-HEXBOLT-1_2-13X1_5 Rev_ New 0 0+ In Work
> (Active) - Creo Parametric]

IS that the PTC_COMMON_NAME that is not changing?  If so - you have to set
that as a parameter value .... I believe.

Dave

adama2000, 2020-07-18:
In the creoson code, there is a section where it sets the drawing's "common
name" to the drawing's root file name -- but ONLY if Creo says that the
common name is modifiable.  Essentially, this:

     if (drw.IsCommonNameModifiable()) { // Creo call
          String commonName = (drawing's file name with the extension
removed)
          drw.SetCommonName(commonName); // Creo call
     }
     else
           (print a line to error output that the common name could not be
modified)

I'm guessing that the "common name" corresponds to PTC_COMMON_NAME, and
maybe something's set so it can't be modified.

Adam Andrews

davidhbigelow, 2020-07-18:
PTC_COMMON_NAME can be set by a parameter set call... I am pretty sure
(assuming that is the issue / solution).

phaleth, 2020-07-18:
For a CAD document the `Name` in Windchill is the `PTC_COMMON_NAME` in Creo Parametric.

adama2000, 2021-01-30:
@victoman1234 did you try setting the PTC_COMMON_NAME parameter on the drawing?

PiNandanwar, 2021-08-11:
I am not the original user who created this ticket, but I also noticed this issue when creating a drawing. 

The one thing I was able to figure out is that if I were to manually create the drawing with the same name, model, and template the PTC_WM_NAME parameter is the same as the drawing.

When I programmatically create the drawing using the Creopyson.drawing.create function, that is when the PTC_WM_NAME variable autofills the template name for that parameter. That parameter is also locked so I cannot just update it with the correct value.