Zepmanbc/creopyson #20: creopyson.parameter.list_ returns exception

stekicar, 2021-01-21:
* Creopyson version: 0.6
* Creoson version: 2.7.0
* Python version: 3.8.7
* Operating System: Win7

### Description

Get param data from part.
If I open part in Creo - code works fine. When part is in memory - code also works fine. If part is not open/active or in memory error pops up: "creopyson.exceptions.MissingKey: Missing `data` in creoson return". Variable "file_" has a value of full path where file is located.
Does part file need to be open or in Creo memory for "creopyson.parameter.list_" to work?

[read_part_param.zip](https://github.com/Zepmanbc/creopyson/files/5845961/read_part_param.zip)

### What I Did

```
file_path = os.path.normpath('C:/TEMP/01_ROBOT_GRIPPER_AND_MAIN_LAYOUT/200844-01-0102_finger.prt')
param_list = creopyson.parameter.list_(c, name="PART_NUMBER", file_=file_path, encoded=None, value=None)

  File "C:\Users\user\AppData\Roaming\Python\Python38\site-packages\creopyson\parameter.py", line 155, in list_
    return client._creoson_post("parameter", "list", data, "paramlist")
  File "C:\Users\user\AppData\Roaming\Python\Python38\site-packages\creopyson\connection.py", line 85, in _creoson_post
    raise MissingKey("Missing `data` in creoson return")
creopyson.exceptions.MissingKey: Missing `data` in creoson return
```

adama2000, 2021-01-21:
The part file **does** need to be in Creo memory.  It **shouldn't** need to be active, as long as you specify the file name as input to the parameter.list command.  I'll check the creoson side.

adama2000, 2021-01-21:
I've tested it with a part that was in Creo memory, but not active, and it did return parameters.

stekicar, 2021-01-21:
Same results here. Thank you for fast reply!

closed 2021-01-21 by stekicar