Zepmanbc/creopyson #21: How to get items from nested family table?

stekicar, 2021-01-25:
Creopyson version: 0.6
Creoson version: 2.7.0
Python version: 3.8.7
Operating System: Win7

### Description

I made family table with nested family tables.
This is hierarchy:
Fastener: SHCS: M3: (3,4,5...), M4: (6,7,8,....), M5: (10,15,20 ...)
Fastener: BHCS: M3: (3,4,5...), M4: (6,7,8,....), M5: (10,15,20 ...)
Fastener: FHCS: M3: (3,4,5...), M4: (6,7,8,....), M5: (10,15,20 ...)

### What I Did

How to get list of items in nested family table for SHCS and the for M3, or M5?

Zepmanbc, 2021-01-25:
Hi
did not tested yet, but I think you must open the parent file of each table to get family information.

Zepmanbc, 2021-01-25:
There is a bug in `c.familytable_list_tree()`
I will correct this asap

```
you should have something like this:
{
    "name":"fastener.prt",
    "total":12,
    "children":[
        "name":"SHCS.prt",
        "total":12,
        "children":[
            "name":"M3.prt",
            "total":12,
            "children":[
                "name":"x3.prt
                "total":0
            ], (....)
        ], (....)
    ], (....)
}
```

Zepmanbc, 2021-01-30:
There is still the question of empty result that will raise an error, but if the part contans a family table it's ok

closed 2021-01-30 by Zepmanbc

stekicar, 2021-02-02:
I tested this new update. Now it works, but I get list at first level, and then string values for list[item] (<class 'str'>). I expected to see another list and then dictionary. Am I missing something?

Zepmanbc, 2021-02-03:
you should get a list of children at first, and then dict in each
show me the result and what you expected

stekicar, 2021-02-03:
Maybe, when I store this data in file I get wrong type on retrieval? I will have to check this.

Zepmanbc, 2021-02-04:
I don't know the reason you store this data in a file but if you don't want to lose time with serialisation and don't need to use it with a third party app, you should use [pickle ](https://docs.python.org/3/library/pickle.html)to keep the types