Zepmanbc/creopyson #22: Opening and adding an instance of a family table

stekicar, 2021-01-27:
Creopyson version: 0.6
Creoson version: 2.7.0
Python version: 3.8.7
Operating System: Win7

### Description
I add an instance of family table in assembly like this:
Although I have dir_name I do not use it and code still works.
`dir_name=Z:\D\0_CREO_TEMP\FINAL
instance_file_name=Z:\D\0_CREO_TEMP\FINAL\DOWEL_6X22<metric_dowel_family_table>.prt
creopyson.file.assemble(c, instance_file_name, dirname=None, generic=None, into_asm=None, path=None, ref_model=None, transform=None, constraints=None, package_assembly=None, walk_children=None, assemble_to_root=None, suppress=None)`

If I want to open instance of family table I will have to use dir_name in code:
`opened_comp = creopyson.file_open(c, instance_file_name, dirname=dir_name, display=False, new_window=True)`
Otherwise it will not work unless working directory is folder where my family table reside. This is expected.

### What I Did
I tried to separate each variable: file_name as instance name, dirname as dir_name and generic as (family table) generic_file_name:
`dir_name=Z:\D\0_CREO_TEMP\FINAL
instance_file_name=DOWEL_6X22.prt
generic_file_name=Z:\D\0_CREO_TEMP\FINAL\metric_dowel_family_table.prt
creopyson.file.assemble(c, instance_file_name, dirname=dir_name, generic=generic_file_name, into_asm=None, path=None, ref_model=None, transform=None, constraints=None, package_assembly=None, walk_children=None, assemble_to_root=None, suppress=None)`
But this does not work for me.
Is this how it should work?
One more thing. How to use variable "constraints" in creopyson.file.assemble? I just put name of component interface but it did not work. Is this used to tell Creo what component interface to use when placing instance in assembly or for something else?
Note: Code formatting does not work properly on github. Everything shows up in one line.
Thank you for your help!

adama2000, 2021-01-27:
The constraints are supposed to be an array of JLConstraint objects - see the creoson functions doc if you have access to it (by default it is at http://localhost:9056/functions.html if creoson is running)

JLConstraint has the following properties:
[image]

An example of specifying a CSYS constraint in JSON (I don't know how you'd specify it in creopyson):
[image]

stekicar, 2021-01-27:
Thank you for reply.
I can make JSON object in python to look exactly like your example. Is there any option to select existing component interface instead? This is because my part is family table and component intercase is already set.

adama2000, 2021-01-27:
I think the source code predates the existence of component interfaces, so
there isn't any provision for them in the code.  But I've never played
around with component interfaces myself.

stekicar, 2021-01-27:
Ok. Thank you for your reply!

Zepmanbc, 2021-01-27:
Hi
you just have to send a dict with the same structure as json

stekicar, 2021-01-27:
Will try! Thank you!

Zepmanbc, 2021-01-27:
I'm not sure I understand all your questiosn but i'll try to answer:
`dir_name` is optional, if you set your working directory it should find the file in it

in order to format your code in you first message you should put:

> \```
> here is your code
> \```

stekicar, 2021-01-27:
What about generic=None option? I thought I need to put generic file name there. When I do that, code does not work.

adama2000, 2021-01-27:
In creoson, the "file" option is supposed to be the instancename+fileext
and the "generic" option is supposed to be the base name.  For example if
you have a part "bracket.prt" which contains a family table, and there is
an entry in the table named "wide_bracket" then you would set
file="wide_bracket.prt" and generic="bracket".  (which may seem backwards).

At least that's how it works for the file-open function.  The file-assemble
should work the same way.

stekicar, 2021-01-27:
Well, it does not. I did what you just explained but if fails. If I put absolute file path for "file", such as for ex. r'Y:\0_CREO\STD_COMPONENTS\FASTENERS\M14X2X20<DIN912_SHCS_BOLT>.prt' and dir_name folder, it works. If I add r'Y:\0_CREO\STD_COMPONENTS\FASTENERS\DIN912_SHCS_BOLT.prt' for generic, for some reason it does not work. Of course, instance name is M14X2X20.prt.

Zepmanbc, 2021-01-27:
if you want test with "vanilla" creoson, you can use `_creoson_post` function

``` c._creoson_post(command, function, data)```

where command should be `file`
function `assemble`
and data is a dict with all params you will find in creoson documentation
something like:
```
{
    "file": instance_file_name,
    "dirname":dir_name,
    "generic":generic_file_name, 
    "into_asm":None,
    "path":None,
    "ref_model":None,
    "transform":None,
    "constraints":None,
    "package_assembly":None,
    "walk_children":None,
    "assemble_to_root":None,
    "suppress":None
}
```

stekicar, 2021-01-27:
Ok. Will do.

Zepmanbc, 2021-02-02:
is it ok?

stekicar, 2021-02-02:
I have not tried "vanilla" version yet.

Zepmanbc, 2021-02-03:
I added  dome doc for vanilla version and logging if it can help
https://creopyson.readthedocs.io/en/latest/usage.html#vanilla-creoson-usage