Zepmanbc/creopyson #23: Creoson code crashes when no material value is set and get_cur_material is used

stekicar, 2021-01-29:
* Creopyson version: 0.6
* Creoson version: 2.7.0
* Python version: 3.9.1
* Operating System: Win10

### Description
Trying to get current material for active part in Creo when there is no set material.
### What I Did
```
cur_material = creopyson.file.get_cur_material(c, file_=None)
```
And this is an error:
```  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO python code\creoson_test.py", line 31, in <module>
    cur_material = creopyson.file.get_cur_material(c, file_=None)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\file.py", line 326, in get_cur_material
    return client._creoson_post("file", "get_cur_material", data, "material")
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\connection.py", line 75, in _creoson_post
    raise RuntimeError(error_msg)
RuntimeError: Error handling request: Unexpected null value```

Is it possible to change creoson code to return empty or "NO_MATERIAL" value so code does not crash if there is no material set in part?

Zepmanbc, 2021-01-30:
This is discuted here [#53](https://github.com/SimplifiedLogic/creoson/issues/53) and here [#59](https://github.com/SimplifiedLogic/creoson/issues/59)

Zepmanbc, 2021-03-22:
fix in Creoson 2.8.0 / Creopyson 0.7.0

closed 2021-03-22 by Zepmanbc