Zepmanbc/creopyson #24: Set_cur_material returns error, but material gets set in model

stekicar, 2021-01-29:
* Creopyson version: 0.6
* Creoson version: 2.7.0
* Python version: 3.9.1
* Operating System: Win10

### Description

Setting new material to model:
```material_name1 = 'AISI_8620_STEEL'
cur_file_name = creopyson.file.set_cur_material(c, material_name1, file_=None)```

### What I Did

Material gets set in model but code fails on return value.

```Current material: AISI_6150_STEEL
Traceback (most recent call last):
  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO python code\creoson_test.py", line 42, in <module>
    cur_file_name = creopyson.file.set_cur_material(c, material_name1, file_=None)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\file.py", line 1083, in set_cur_material
    return client._creoson_post("file", "set_cur_material", data, "files")
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\connection.py", line 85, in _creoson_post
    raise MissingKey("Missing `data` in creoson return")
creopyson.exceptions.MissingKey: Missing `data` in creoson return```

I use "try:" to avoid breaking the code for now,

adama2000, 2021-01-29:
This is a bug in the creoson layer, it's not returning the data it's supposed to for set_cur_material.  I'll fix that.

adama2000, 2021-01-30:
Checked in a source-code fix to creoson.  This will be included in the next Release.

closed 2021-01-30 by Zepmanbc