Zepmanbc/creopyson #25: creopyson.drawing.create_symbol fails with RuntimeError: No 'replace_values' parameter given

stekicar, 2021-01-29:
* Creopyson version: 0.6
* Creoson version: 2.7.0
* Python version: 3.9.1
* Operating System: Win10

### Description
Placing symbol on drawing. Symbol itself has value that changes based on model parameter value.
```
symbol_name = 'APPROVAL_STAMP'    
sym_loc = {"x": 424.8, "z": 0.0, "y": 273.4}
creopyson.drawing.create_symbol(c, symbol_name, sym_loc, drawing=None, replace_values=None, sheet=None)
```

### What I Did
RunimeError occurs. replace_values should be optional.
```
Traceback (most recent call last):
  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO python code\drawing_add_approval_symbol.py", line 56, in <module>
    creopyson.drawing.create_symbol(c, symbol_name, sym_loc, drawing=None, replace_values=None, sheet=None)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\drawing.py", line 247, in create_symbol
    return client._creoson_post("drawing", "create_symbol", data)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\connection.py", line 75, in _creoson_post
    raise RuntimeError(error_msg)
RuntimeError: No 'replace_values' parameter given
```
Is this a bug or I am doing something wrong?

adama2000, 2021-01-29:
You're right, that's a bug.  The  replace_values option is supposed to be optional, but the creoson code is requiring it.  I'll fix it.

adama2000, 2021-01-30:
Checked in a source-code fix to creoson.  This will be included in the next Release.

closed 2021-01-30 by Zepmanbc