Zepmanbc/creopyson #26: creopyson.feature.list_ and creopyson.feature.list_params return

stekicar, 2021-02-04:
* Creo version: 5.0
* Creopyson version: 2.7.0
* Creoson version: 0.6.1
* Python version: 3.9.1
* Operating System: Win7

### Description

I have a part with couple of different holes:
[image]
And I am trying to get a list of all features.

### What I Did

```
try:
    part_features1 = creopyson.feature.list_params(c, file_=None, name=None, type_=None, no_datum=None, inc_unnamed=None, no_comp=None, param=None, value=None, encoded=None)
    part_features2 = creopyson.feature.list_(c, file_=None, name=None, status=None, type_=None, paths=None, no_datum=None, inc_unnamed=None, no_comp=None)
except:
    print('Fails!'')
print('part_features1:', part_features)
print('part_features2:', part_features)
```
I do not run both at the same time. When I run part_features1, try: fails. When I run part_features2, I get list but only datums' info is in the list. No holes info:
```
part_features2: [{'status': 'ACTIVE', 'name': 'DEFAULT_CSYS', 'type': 'COORDINATE SYSTEM', 'feat_id': 1}, {'status': 'ACTIVE', 'name': 'YZ_RIGHT', 'type': 'DATUM PLANE', 'feat_id': 3}, {'status': 'ACTIVE', 'name': 'XZ_FRONT', 'type': 'DATUM PLANE', 'feat_id': 5}, {'status': 'ACTIVE', 'name': 'XY_TOP', 'type': 'DATUM PLANE', 'feat_id': 7}]
```
I also tried to group all holes into 'AAA' groups and then I get all holes in the list:
```
part_features: [{'status': 'ACTIVE', 'feat_number': 8, 'type': 'HOLE', 'feat_id': 255}, {'status': 'ACTIVE', 'feat_number': 9, 'type': 'HOLE', 'feat_id': 349}, {'status': 'ACTIVE', 'feat_number':
10, 'type': 'HOLE', 'feat_id': 446}, {'status': 'ACTIVE', 'feat_number': 11, 'type': 'HOLE', 'feat_id': 495}, {'status': 'ACTIVE', 'feat_number': 12, 'type': 'HOLE', 'feat_id': 563}]
```

What should I do for creopyson.feature.list_ to get list of all features? Also, why "creopyson.feature.list_params" fails?
Thank you!

stekicar, 2021-02-04:
I also run: creopyson.feature.list_ and group 'AAA' shows up in the list which makes sense because holes are now hidden in the model tree (in case hidden is meant for visibility within model tree and not within model view):
```
part_features: [{'status': 'ACTIVE', 'name': 'DEFAULT_CSYS', 'type': 'COORDINATE SYSTEM', 'feat_id': 1}, {'status': 'ACTIVE', 'name': 'YZ_RIGHT', 'type': 'DATUM PLANE', 'feat_id': 3}, {'status': 'ACTIVE', 'name': 'XZ_FRONT', 'type': 'DATUM PLANE', 'feat_id': 5}, {'status': 'ACTIVE', 'name': 'XY_TOP', 'type': 'DATUM PLANE', 'feat_id': 7}, {'status': 'ACTIVE', 'name': 'AAA', 'type': 'GROUP HEAD', 'feat_id': 623}]
```

adama2000, 2021-02-04:
Are the Hole features named?  If not, try setting the "inc_unnamed" option
to True -- this will include un-named features.

stekicar, 2021-02-04:
Yes. That's it. Now it works! Thank you!
Why "inc_unnamed" should be true when all of my features have names in model tree?
[image]

stekicar, 2021-02-04:
Also, what creoson command can I use to pull notes from feature if any? I can get all notes but how to link which note belongs to what feature (in my case: hole notes). I was expecting to see feature_id with creopyson.note.list_ output:
```
{'name': 'Note_4', 'value': "C'BORE-{1:&PATTERN_NO}X\nFOR &FASTENER_ID\n", 'encoded': False}
```

adama2000, 2021-02-04:
Names like "Hole 1" (feature type + index) are not real names, they're just
placeholders that are automatically generated on the Creo screen but are
not stored internally in Creo.

Creoson doesn't have a way of showing feature/note relationships now, maybe
that's something to look into.

stekicar, 2021-02-04:
I am trying to make hole table more usable in Creo. It is a joke that we cannot use notes as parameters in Creo hole tables. This should have been default since Pro/E version 1! This is the reason I want to correlate notes to holes and attach note data to hole feature parameter and then call that parameter in Creo hole table.
Thank you for your prompt answers!

stekicar, 2021-02-04:
Is there a way to get all dimension per feature?

adama2000, 2021-02-04:
Not within Creoson.  I'll have to research to see whether JLink supports it.

Zepmanbc, 2021-03-22:
can be closed?
fix in Creoson 2.8.0 / Creopyson 0.7.0 ?

adama2000, 2021-03-22:
Not yet.  I never researched that last question.

adama2000, 2021-03-23:
@stekicar I'm not sure that JLink can support listing dimensions for a feature, but it's been decided that we're not going to pursue the ability.

stekicar, 2021-03-23:
Ok.

closed 2021-03-23 by Zepmanbc