Zepmanbc/creopyson #28: creopyson.exceptions.MissingKey: Missing `data` in creoson return

stekicar, 2021-02-08:
* Creo version: 5.0.06
* Creopyson version: 0.6.1
* Creoson version: 2.7.0
* Python version: 3.9.1
* Operating System: Win10

### Description

Trying to set material in active mode:
```
creopyson.file.set_cur_material(c, material, file_=None)
```

### What I Did
Returns an exception:
```
Traceback (most recent call last):
  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO_python_code\set_material_to_active_model.py", line 58, in <module>
    setMaterial('ALUM_6061-T6', r'Y:\0_CREO\MATERIALS')
  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO_python_code\set_material_to_active_model.py", line 33, in setMaterial
    creopyson.file.set_cur_material(c, material, file_=None)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\file.py", line 1083, in set_cur_material
    return client._creoson_post("file", "set_cur_material", data, "files")
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\connection.py", line 85, in _creoson_post
    raise MissingKey("Missing `data` in creoson return")
creopyson.exceptions.MissingKey: Missing `data` in creoson return
```
I guess, its a bug?

Zepmanbc, 2021-02-09:
can you use vanilla Creoson or debug to see what are the query and response to Creoson please?

adama2000, 2021-02-09:
I'll note that set_cur_material is one of the functions that will not return a data object if there are no matching files to set the current material on -- and if he's saying "file_=None" then maybe there isn't an active model in Creo?

adama2000, 2021-02-09:
Are you trying to set a material that hasn't yet been loaded into the models you're setting it on?

stekicar, 2021-02-10:
No. First I check if material is loaded and if not, I load material file first and then assign material. Problem is that material is assigned  to model but error exception occurs anyway. So, I have to use try: except: in order to avoid code error. But the problem is that then I do not know of/when real error occurs.

stekicar, 2021-02-10:
> can you use vanilla Creoson or debug to see what are the query and response to Creoson please?

Can you tell me how to do this?

Zepmanbc, 2021-02-10:
https://creopyson.readthedocs.io/en/latest/usage.html#vanilla-creoson-usage

tell me if you need more help
with vanilla mode there is no *data* key waited in the response so you will have full Creoson response

stekicar, 2021-02-11:
I run vanilla code as:
```
command = 'file'
function = 'set_cur_material'
data = {'file':'200884-01-0202_stand.prt', 'material':'AISI_1045_STEEL'}
result = c._creoson_post(command, function, data)
print('result:', result)
```
No exception occurs and material gets assigned. But result returns None data.
[image]

Zepmanbc, 2021-02-12:
Thanks
So it works but `data` is missing.
You need to wait Creoson update (or you can build by yourself with last commit, I think adama2000 pushed the modifications)
Or add try/except in your code

adama2000, 2021-02-12:
I do plan to do a new Creoson release as soon as it seems like as many
tickets have been addressed as can be.

Adam

Zepmanbc, 2021-03-22:
fix in Creoson 2.8.0 / Creopyson 0.7.0

closed 2021-03-22 by Zepmanbc