Zepmanbc/creopyson #29: Not an issue - Just a question

stekicar, 2021-02-21:
* Creo version: 5.0.6
* Creopyson version: 0.6.2
* Creoson version: 2.7.0
* Python version: 3.9.1
* Operating System: Win10

### Description

Does creopyson/creoson have module that collects coordinates of clicked mouse position in drawing? If not, how hard could it be to make one?

Zepmanbc, 2021-02-21:
Hello, I don't think it is possible through Creoson
the only feature with user interaction I know is `feature.user_select_csys`

To get mouse position you can use [PyAutoGui](https://pyautogui.readthedocs.io/en/latest/mouse.html) but I don't kwnow if it will fit with your needs

stekicar, 2021-02-21:
Ok. I need something like this to place symbol in drawing manually. I will see if PyAutoGui will do the trick.

Zepmanbc, 2021-02-21:
If you want to pause your script to put some symbols manualy, you can easily put a message box with [PyautoGui](https://pyautogui.readthedocs.io/en/latest/msgbox.html) too, and then resume the script when it's done.

stekicar, 2021-02-21:
"creopyson.drawing.create_symbol" needs symbol location parameter and I will have to see how pixel location translate to creo drawing location. Especially when zoomed in.

Zepmanbc, 2021-02-21:
you can run a mapkey to resize screen

adama2000, 2021-02-21:
I don't think JLink in general provides the ability to address mouse positions or screen coordinates.  So it's unlikely creoson could support it.

stekicar, 2021-02-22:
Can JLink return selected features or surfaces? Or anything else that is selected? I know for dimensions and CSYS.

adama2000, 2021-02-22:
Funny you should ask, I just an hour ago checked in a new
feature.list_selected function to return what features are selected.  It
will be in the next release.

Creoson issue: https://github.com/SimplifiedLogic/creoson/issues/32

davidhbigelow, 2021-02-22:
BUT WHY?

What will be the purpose of this after the selection?

stekicar, 2021-02-22:
While I was working in Inventor I was heavily using VBA to speed up processes and minimize clicks. I was able to build a code that let you select round surface of the hole (or axis) and flat surface, then code would calculate which McMaster fastener to place, it would auto constrain and pattern fastener if hole was patterned. I was using measurement tool in code to get all data I need to pick and place fasteners.
So, I would like to do the same thing with Creo.

davidhbigelow, 2021-02-22:
Seems ambitious for Creo...

stekicar, 2021-02-22:
I did not try VBA with Creo yet, just because it looks pretty cumbersome - no VBA IDE built in Creo. And who knows what Creo API currently supports in code. That is why creoson/creopyson looked to me as much better solution.

davidhbigelow, 2021-02-22:
CREO does have a VB interface — my understanding it is less capable thank
JLINK .... not a VB guy, I try to avoid it. (Painful to debug)

The selecting of a feature is one thing ... you can do things like suppress
or delete by Feature Id...

Surface ID is not as directly actionable ...  but might be useful within a
mapkey?!? (Maybe) and not directly related to a specific feature...
(imported geometry would not enable a derived relationship to a protrusion,
cut or fillet)

Creoson, while it does have some lower level capabilities is not really
intended to augment user mouse picks...  the goal is to not require much
from user interactions.

This is why my question was raised...  the concern that even if we could
get some information.... what would you expect to do next...  features are
one thing, surfaces are another ... it can get pretty complicated after
that - often bumping into JLINK (even Object TookKit) limitations pretty
quickly depending on what your next steps and goals are...

The risk — we add something in that results in a hard  “dead end” for what
you can do after.

stekicar, 2021-02-22:
Are you saying that Creo has VBA IDE built in like Excel or Inventor have?
I do understand your concern about dead ends. Creo is full of surprises ... Look at how Creo manages hole tables, total disaster.

davidhbigelow, 2021-02-22:
Yea... I am saying that Creo has a VB interface...  given that you have VB
experience...  I would be curious as to your view/experience.

Hole tables ...  particularly difficult.   Mapkeys make them better....
we added special functionality in Nitro-CELL to support the writing of the
Pro/TABLE format from Excel...  but it still needs a mapkey to
integrate....  pretty sure Object Toolkit is still painful for that (I am
sure if anyone disagrees with that statement ...  they will).

closed 2021-09-01 by Zepmanbc