Zepmanbc/creopyson #30: creopyson.note.delete() force close connection to Creo

gazontimide, 2021-03-15:
* Creo version: 3
* Creopyson version: 0.6.2
* Creoson version: 2.7
* Python version: 3.7
* Operating System: windows 10

### Description

When trying to use the function to delete a note from a drawing, it shuts down my connection to Creo and throws an error.

### What I Did

Simply use the creopyson.note.delete() function to delete a specific note from a drawing.

```
Paste the command(s) you ran and the output.

clientobj = creopyson.connection.Client()
clientobj.connect()
creopyson.note.delete(clientobj, name='Note_75')

If there was a crash, please include the traceback here.

Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1321, in getresponse
    response.begin()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 296, in begin
    version, status, reason = self._read_status()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 257, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\util\retry.py", line 400, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1321, in getresponse
    response.begin()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 296, in begin
    version, status, reason = self._read_status()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 257, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 53, in _creoson_post
    r = requests.post(self.server, data=json.dumps(request))
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "U:/Creo application ingenierie/testing.py", line 20, in <module>
    creopyson.note.delete(clientobj, name='Note_75')
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\note.py", line 63, in delete
    return client._creoson_post("note", "delete", data)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 55, in _creoson_post
    raise ConnectionError(e)
ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

Process finished with exit code 1

```

Zepmanbc, 2021-03-15:
Hi, I cannot test right now but is there a message in Creoson window? (I suppose you start Creson with `creoson_run.bat` file)

gazontimide, 2021-03-15:
This is the error message in the console window when I start Creoson with the .bat file : 

# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ffb9f3e9c60, pid=69156, tid=71224
#
# JRE version: Java(TM) SE Runtime Environment (7.0_65-b19) (build 1.7.0_65-b19)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.65-b04 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [pfcasyncmt.dll+0x179c60]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# T:\Pro_stds\Creo\SMLCreoApp\hs_err_pid69156.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.

I also attached the log file if that can help.

[hs_err_pid69156.log](https://github.com/Zepmanbc/creopyson/files/6143775/hs_err_pid69156.log)

adama2000, 2021-03-15:
Okay, the JVM is crashing in "pfcasyncmt.dll", which is PTC's JLink
interface DLL.  This means that the crash is occurring in PTC's library,
not in creoson or creopyson.

The trace log you attached shows that the crash occurs when creoson is
trying to find your note by looping through the notes in your drawing and
checking the note name to see if it matches the name you passed in.

Do you get the same error if you call notes.list?  You probably do, for the
same reason.

I suspect there is something "unusual" about one or more of the notes in
your drawing that causes the crash -- it may not even be the note you're
trying to delete, since the error occurs when it's trying to loop through
all the notes.

If you can post a sample drawing/model here which duplicates the problem,
we might be able to diagnose it further.  I couldn't duplicate the issue
with one of our drawings.

adama2000, 2021-03-15:
One thing I wonder is if one or more of the notes in your drawing doesn't *have* a name, and it crashes when the loop hits that note.

adama2000, 2021-03-15:
Hm, I read something wrong -- it's NOT looping through all the notes, it's trying to get your specific note by name (getItemByName()).  So it should be this *specific* note that has the problem.

gazontimide, 2021-03-16:
I'll be honest, there are some things I never quite understood with a drawing's note definition. I know there's 2 layers which contains note ids, "DETAIL_ITEM" and "NOTE" which are both standard in a Creo drawing file (I think). I don't know if having the same note ids in 2 different layers make it crash or something?

What's also weird is that the command creopyson.note.list_() well returns a list with all the drawing notes.

I did some testing with a blank drawing using an empty format. I added a manual note "test". First I call the note.list_() method to get the note's name and then when I try to delete it using the note.delete() method it crashes.

Code : 
clientobj = creopyson.connection.Client()
clientobj.connect()
print(creopyson.note.list_(clientobj))

output : 
[{'location': {'x': 240.98708935611035, 'z': 0.0, 'y': 234.17711070959268}, 'name': 'Note_0', 'value': 'test', 'encoded': False}]

Process finished with exit code 0

adama2000, 2021-03-16:
Hm, maybe the reason is that when we're trying to delete a note, we try to
get the note by its name first -- and if there are multiple notes with the
same name, it causes a crash?

What happens if you specify a "name" property (setting it to "test") when
you call note.list?

gazontimide, 2021-03-16:
Using the value of the note to try and delete it does not work either and interestingly enough, no matter what I type in the "Name" parameter, it crashes.

`Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1321, in getresponse
    response.begin()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 296, in begin
    version, status, reason = self._read_status()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 257, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\util\retry.py", line 400, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\urllib3\connectionpool.py", line 416, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1321, in getresponse
    response.begin()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 296, in begin
    version, status, reason = self._read_status()
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 257, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 53, in _creoson_post
    r = requests.post(self.server, data=json.dumps(request))
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "U:/Creo application ingenierie/testing.py", line 19, in <module>
    creopyson.note.delete(clientobj, name='blabla')
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\note.py", line 63, in delete
    return client._creoson_post("note", "delete", data)
  File "C:\Users\samgir\AppData\Local\Programs\Python\Python37\lib\site-packages\creopyson\connection.py", line 55, in _creoson_post
    raise ConnectionError(e)
ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

Process finished with exit code 1
`

adama2000, 2021-03-16:
> File "", line 3, in raise_from

Any chance of you posting a sample drawing/model here so we can duplicate
the issue and try to see if there's a workaround?

gazontimide, 2021-03-16:
Yeah, sure. Very simple drawing.
[test_note_function.zip](https://github.com/Zepmanbc/creopyson/files/6152180/test_note_function.zip)

adama2000, 2021-03-16:
Hm, I got a crash but not the same one you did.  I'll investigate further
but may have to open a ticket with PTC.  What version of Creo are you
running?

adama2000, 2021-03-16:
Never mind, I see you said Creo 3.

adama2000, 2021-03-18:
Can you tell me the Creo 3 Datecode?  (M100, M190, or what?)

gazontimide, 2021-03-18:
Old M030!