Zepmanbc/creopyson #33: New Creoson release

adama2000, 2021-03-21:
Having just published a new creoson release, here's the changes that I think require creopyson updates:

New functions:

- creo.set_creo_version - set the Creo version (4,5,6,7,etc) as a fix for Creo 7 for regeneration handling.
	takes a "version" integer argument, which is required
	returns nothing

- feature.list_selected - list the currently selected features in Creo.
	takes no arguments
	returns a list of objects with the following contents:
		file - model name (string)
		name - feature name (string)
		status - feature status (string)
		type - feature type (string)
		feat_id - feature ID (int)
		feat_number - feature number (int)
		path - feature's component path (array of ints)

Modified functions:

- drawing.list_view_details
	add "view_model" return value to objects in return array
	add "simp_rep" return value to objects in return array

- feature.list_params
	add "description" return value to objects in return array

- feature.set_param
	add optional "description" string argument

- feature.user_select_csys
	add "file" return value to objects in return array
	remove "feat_number" return value from objects in return array

- file.set_cur_material
	fix bug so it returns a list of the file names affected (string array)

- parameter.list
	add "description" return value to objects in return array

- parameter.set
	add optional "description" string argument

Zepmanbc, 2021-03-22:
@adama2000 
object-FeatSelectData.json: there is still *feat_number* in the doc

Zepmanbc, 2021-03-22:
V0.7.0 online

closed 2021-03-22 by Zepmanbc