Zepmanbc/creopyson #39: how to download or open data (list of parts) from Windchill

tcrakshi, 2021-07-15:
* Creo version:
* Creopyson version:
* Creoson version:
* Python version:
* Operating System:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

tcrakshi, 2021-07-15:
some use case / example code is much appreciable

Zepmanbc, 2021-07-16:
```
import creopyson

c = creopyson.Client()
c.connect()

# Here I suppose your workspace is activated
list_of_files = c.windchill_list_workspace_files()
```
https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.windchill.list_workspace_files

tcrakshi, 2021-07-16:
Hi, 

Thanks for update, 

1. If part is not available in my local  workspace, then I need to search in server, how we can do that? 

2. If part search is succeeded, then how I can download the data into specific local folder? Or how part will open in creo? , Some use case example code is much appreciable.

Zepmanbc, 2021-07-16:
If part is not available in your local workspace, you can add it if you know its exact name.
If you just want to add it to your workspace and don't open it in creo you can add *display* and *activate* args
```c.file_open("my_file_name.prt", display=False, activate=False)```
This method does not work 100% with recents files, I don't have understand yet if there is a cache not updated for new files or if it is my job's windchill's configuration.

Like said before, you must activate your local workspace in order to add files in it.
If you want add differents files to differents workspaces, you will have to switch between your workspaces.
You can find everything available in Creoson for Workspace management here: https://creopyson.readthedocs.io/en/latest/creopyson.html#module-creopyson.windchill

tcrakshi, 2021-07-19:
I am getting error in c.connect()

my Code
'''
import creopyson

c = creopyson.Client()
c.connect()  # Errro geeting in this line

# Here I suppose your workspace is activated
list_of_files = c.windchill_list_workspace_files()
'''

**Error**
'''
    raise ConnectionError(e)

ConnectionError: HTTPConnectionPool(host='localhost', port=9056): Max retries exceeded with url: /creoson (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x000001BD1C2F2448>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it'))

'''

Zepmanbc, 2021-07-19:
Creo and Creoson are running? (only 1 instance of each)

tcrakshi, 2021-07-19:
creo is running, how to check Creoson is running or not?
just i am running python code. is there anything i need to predefined before running python code?

Zepmanbc, 2021-07-19:
Did you launch Creson? This is the best way to know if it's running.
after configuration you must click on «Start CREOSON»

https://creopyson.readthedocs.io/en/latest/usage.html#quickstart
https://www.youtube.com/watch?v=NjkvRZJQzXs

closed 2024-11-22 by Zepmanbc