Zepmanbc/creopyson #41: How to print drawings and save them as .dwg

Y2KBIOHAZARD, 2021-09-25:
* Creo version: 3.0
* Creopyson version: 0.7.3
* Creoson version:
* Python version: 3.8
* Operating System: win 10

### Description

Trying to print all drawings from working directory and save them as .dwg
- I have started creo and creoson  
- make folder in working directory where I intent to store .dwg files 
- open every .drw file and get specific parameter 
- open every sheet and get format that drawing is using

I'm stuck here and don't know how to:
- save file as .dwg 
- Get the print preview in creo
- Print file on different printer depending of format drawing is using

Thank you for your help

### What I Did

```

import creopyson
import os

c = creopyson.Client()
c.connect()

print("Creo is running: ", c.is_creo_running())
current_directory = c.creo_pwd() #  return current working directory.
listfiles = c.creo_list_files("*.drw") #  return a list in the working directory.
print("list of drw files: ", listfiles)
pateka = os.path.join(f"{current_directory}Autocad")  #  create and use Autocad folder in WD
if not os.path.isdir(pateka):
    os.makedirs(pateka)

for file in listfiles:
    c.file_exists(file)  #  verify if "file" exists.
    c.file_open(file, display=True)  #  Open "file" in Creo.
    c.drawing_regenerate(file)  #  regenerate drawing
    drawing_sheets = c.drawing_get_num_sheets(file)  #  get number of sheets in drawing
    print("opening file: ", file,"and have number of sheets: ",drawing_sheets)

    i=1
    while i<=  drawing_sheets: 
        sheet_format = c.drawing_get_sheet_format(sheet=i)
        parametar_pozcija = c.parameter_list("poz")
        parametar_pozcija = list(parametar_pozcija)

        new_list = list(sheet_format.values())
        if any ("A4" in word for word in new_list):
            format_crtez = "A4"
            printer_name = r"\\05OHPC\hp LaserJet 1320PCL 5" #test printer
            print("format of drawing: ",format_crtez, "sheet: ", i)
        elif any ("A3" in word for word in new_list):
            format_crtez = "A3"
            print("format of drawing: ",format_crtez, "sheet: ", i)
        elif any ("A2" in word for word in new_list):
            format_crtez = "A2"
            print("format of drawing: ",format_crtez, "sheet: ", i)
        elif any ("A1" in word for word in new_list):
            format_crtez = "A1"
            print("format of drawing: ",format_crtez, "sheet: ", i)
        elif any ("A0" in word for word in new_list):
            format_crtez = "A0"
            print("format of drawing: ",format_crtez, "sheet: ", i)
        else:
            print("nonstandard format: ", file, "sheet: ", i)

        file_name = f"{file}_{i}"
        file_autocad = f""{parametar_pozcija}"{file}_{i}" # filename for .dwg
        i=i+1
    c.file_close_window(file)
    print("closing file: ", file)

c.file_erase_not_displayed()
```

Zepmanbc, 2021-09-25:
Hello
I think the best way is to create mapkeys for each printer and you run it with [interface_mapkey](https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.interface.mapkey).
Same thing for DWG export, there is no build-in feature for this export format in Creoson.

If you need to check something before run the print or validate something, personally I use [pyautogui's message box](https://pyautogui.readthedocs.io/en/latest/msgbox.html). I don't know if it this concern your case, maybe the preview (that should be launch with a mapkey too)?

Just an advice, you should use [pathlib](https://docs.python.org/3/library/pathlib.html) instead of `os.path`, it works very well and prevent from a lot of problems with Windows file system, naming, absolute/relative path, etc...

Y2KBIOHAZARD, 2021-11-04:
Hello,
Thank you for the advice. I'm beginner in programing, so any advice is welcome.

Didn't manage to print directly. I had problem setting the printers in creo. But I succeed to export all drawings from working directory/ workspace as dwg and pdf (with mapkey method), then separate pdfs by drawing format in different folders(A0,A1...) and then merge them into one file. (There was an error with large number files while merging with PyPDF2, needed to be added this code and problem was solved.)
```
import os
sys.setrecursionlimit(150000)
```

This is in working directory
[image]

This is in PDF_PRINT folder and only need to print those files
[image]

I'm trying to make another program for adding relations in all parts from working directory/ workspace and noticed that relations are added, but don't take effect unless change something in relations manually or click verify. Also all pre-existing relations are erased.

One more question, how can I start the batch file "creoson_run.bat" from python, I tried with:
```
import subprocess
subprocess.call(['C:\path\creoson_run.bat'])
```
and get this message:
"The setvars.bat file does not exist, run CreosonSetup.exe to create it."
when started manually don't have problem.

Thank you.

Zepmanbc, 2021-11-04:
To start Creoson from your script you must run 1 time Creoson with **CreosonSetup.exe**, configure your Creo path and click *Start*.
It will create the **creoson.settings** file, in the Creoson folder, with your configuration *(the port number and the path to Creo)*.
Then you should start Creoson by launching **creoson_run.bat**

I think the "subprocess.call" stuff does not work, did you find this in the documentation? I did not remember where it is...

This should works:
```
import os
from pathlib import Path

CREOSON_PATH = r"C:/Creoson/path/"

os.chdir(CREOSON_PATH)
run_path = Path(CREOSON_PATH ) / "creoson_run.bat"
os.startfile(run_path.resolve().as_posix())
```
`os.chdir` set the current directory
`run_path` is a Path object that allow to use these methods: `.resolve().as_posix()` that write properly the path (not the shitty windows way...)
`os.startfile` launch the batch file

https://docs.python.org/3/library/pathlib.html#pathlib.PurePath.as_posix
https://docs.python.org/3/library/pathlib.html#pathlib.Path.resolve

closed 2024-11-22 by Zepmanbc