Zepmanbc/creopyson #42: RuntimeError: The file 'abc' could not be opened

akash9470, 2021-11-01:
* Creo version: 7
* Creopyson version:
* Creoson version:
* Python version:3.7
* Operating System:w10

### Description

i am trying to open step file in creo using following method

INPUT-
import creopyson
c = creopyson.Client()
c.connect()
c.creo_set_creo_version(7)
c.creo_cd("C:\\Users\\Desktop\\New")
c.interface_import_file("abc","STEP",)

OUTPUT-

  File "C:\New folder\untitled1.py", line 13, in <module>
    c.interface_import_file("abc","STEP",)

  File "C:\Program Files\Spyder\pkgs\creopyson\interface.py", line 363, in import_file
    return client._creoson_post("interface", "import_file", data, "file")

  File "C:\Program Files\Spyder\pkgs\creopyson\connection.py", line 75, in _creoson_post
    raise RuntimeError(error_msg)

RuntimeError: The file 'abc' could not be opened

Zepmanbc, 2021-11-02:
Hi @akash9470 
you need to name the file with his extension:
```
import creopyson
c = creopyson.Client()
c.connect()
c.creo_set_creo_version(7)
c.creo_cd("C:\Users\Desktop\New")
created_file = c.interface_import_file("abc.stp")
c.file_open(created_file)
```
the *created_file* will be an asm if you don't force **new_model_type** arg to *prt*
```created_file = c.interface_import_file("abc.stp", new_model_type="prt")```

akash9470, 2021-11-03:
Thank for the help....it worked

closed 2021-11-03 by Zepmanbc