Zepmanbc/creopyson #43: Using creopyson/creoson to build an assembly??

danachee, 2022-01-26:
* Creo version: 3.0
* Creopyson version: 0.7.3
* Creoson version: 2.8.0
* Python version: 3.8.10
* Operating System: Linux (Ubuntu)

### Description

This is a question, not a bug.  I can't find any examples of how, if its possible, to build an assembly using only creopyson.  For example, let's say I have two blocks and I want to mate them face to face, with a space between.   How would I go about this?

Thanks,

### What I Did

Googled "using c.file_assembly()" in various combinations

```
```

davidhbigelow, 2022-01-26:
Dana,

There are a few ways to do this - Coordinate Systems is the preferred way
for consistency in any automation.

It is possible to generically constrain things like a user would - but this
requires more planning to be useful more generically.

in the CREOPYSON docs : look at file_assemble - and the constrains portion
as a reference.

https://creopyson.readthedocs.io/en/latest/creopyson.html?highlight=assemble#creopyson.file.assemble

This is a quick generic example from the main CREOSON reference :
http://www.creoson.com/functions.html

look up:
file : assemble
object : JLConstraint

You should find similar in the CREOPYSON docs.

Dave

danachee, 2022-01-26:
Thanks!   Is there a way to pull this info from an existing assembly or component?  That way I'll have a proper way of matching up what exactly should be in the JLConstraint?

davidhbigelow, 2022-01-26:
Not sure that is 100% possible - may require some listing of feature types
- which should be by "name" ... so I would not recommend that approach.

Best to plan for the structure up-front and pick the most stable references
that are consistent.   Coordinate Systems always work and are less overhead
to manage.

Just my $0.02 on this.

Dave

danachee, 2022-01-26:
Thanks!  (I'm trying to code this up for someone else.   Mechanical engineering isn't my skillset!).  I'll play with it to figure it out.   (but a quick question, which part of the API would be relevant to Coordinate Systems usage?).

davidhbigelow, 2022-01-26:
Not sure in CREOPYSON specifically - but it is a parallel to the standard
CREOSON functionality.

http://www.creoson.com/functions.html

Search for "file : assemble" --- there are a few examples there for using
Coordinate Systems as constraints - should be simple to implement using
Coordinate System Names.

Note - the files have to be in the Creo session memory before doing this...
and the assembly as to be active you are assembling to.

Hope this helps.

Dave

danachee, 2022-01-26:
Ah, Thank you so much!   I see that the 'csys' name for the 'type' is the link.   Feel free to close this out.