Zepmanbc/creopyson #44: Automate drawings of family table instances

RandomFab, 2022-02-05:
* Creo version: 7
* Creopyson version: 
* Creoson version:
* Python version: 3.9
* Operating System: windows

### Description
I'm definetly not an expert in python and coding in general, so please be indulgent if i'm doing BIG mistakes !
Description ->
I have designed a generic part and have created its family table on creo. I have done a drawing based on the generic part. I want to generate a drawing for every single instance of my generic part.

On creo you can manually "replace" the drawing model by the instance that you want. I basically try to do the same with creopyson.

### What I Did

I tried : 
>>> c.drawing_set_cur_model("part0001_inst2.prt")
[...]
    RuntimeError: File 'part0001_inst2.prt' was not open

>>> c.drawing_set_cur_model("PRT0001_INST2<PRT0001>.prt")
[...]
RuntimeError: A Pro/TOOLKIT error has occurred: Not Found.  Check your model/drawing for accuracy.

I'm not sure about what i'm doing rn. So please, if you have any idea or advices, do not hesitate !

davidhbigelow, 2022-02-05:
You might consider using a drawing template as your startpoint.

This is a good practice in general - especially when trying to get a bunch
of initial drawings created for a stack of models (e.g. loop through the
models and execute create a drawing using a template reference).

drawing : create using the template name as an argument.

Dave

Zepmanbc, 2022-02-08:
Hello @RandomFab 
@davidhbigelow is right, template is the best practice.
But I think t\you must be very strict with your model to use «build dimensions» (sorry don't know the exact word in english) to auto complete the drw.
You should try to do a mapkey that changes your model in the drawing.
After that you can change the code with the list of names in your family table.
to use mapkey: https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.interface_mapkey

closed 2024-11-22 by Zepmanbc