Zepmanbc/creopyson #45: geometry_bound_box also measures sketches

leppeles, 2022-03-09:
* Creo version: tested on 4 M110 and 7.0.2
* Creopyson version: 0.7.2
* Creoson version: 2.8
* Python version: 3.9
* Operating System: Windows 10

### Description

Boundary box sizes from creopyson contains sketch sizes also, not just geometrical elements.
At sheetmetal parts it is common practice to draw a sketch first as the profile, then extrude it and create the flat. The flat also contains the initial sketch and its "height" boundary box measurement will not be the thickness of the part, but the height of the sketch.

### What I Did

I was trying to extract the boundary box sizes of a list of parts. Turns out the boundary box also includes sketches, which is not the actual geometry. Especially occurs at the sheetmetal parts as I wanted to know their flats' boundary box sizes. As on the attached image, Z size is not the flat's thickness but the sketch's height.

[image]

Zepmanbc, 2022-03-10:
Hi @leppeles 
Does the blue lines' height is 200?
Boundary Box takes everything in the model

leppeles, 2022-03-10:
Hi @Zepmanbc 
Yes, blue line's height is 200, therefore I cant extract the real geometry's boundary box sizes.

Zepmanbc, 2022-03-10:
No, not with boudary box, it's only for global model.
But for your problem, you define the thickness so it's the only dimensions you cannot have with the boundary box, the X/Y are OK

leppeles, 2022-03-10:
That is true, however I can not be sure that the Z will be the thickness every time, without looking the model. 
I only see that 3 values are extracted as X/Y/Z lets say 50/60/70 and know that it is a sheetmetal because earlier I ask for instances and if it has a flat, I ask for the flat's boundary box. But even if I get the thickness, I wouldn't know which of these is wrong.

So I figured that its easier if I just use a different approach for sheetmetals and extract the following parameter values:
SMT_THICKNESS with parameter_list(name="SMT_THICKNESS")
SMT_FLAT_PATTERN_LENGTH with feature_list_params
SMT_FLAT_PATTERN_WIDTH with feature_list_params

And this should solve my issue.

closed 2022-03-16 by leppeles