Zepmanbc/creopyson #46: Handling creopyson.exceptions

stekicar, 2022-03-14:
* Creo version: 5.0.6
* Creopyson version: 0.7.2
* Creoson version: 2.7.0
* Python version: 3.10
* Operating System: Win10

### Description

How to handle creopyson.exceptions the right way? I tried to raise an error, but program crashes instead of catching error.

### What I Did

```
    file_name = 'test'
    dir_name = 'test'
    try:
        creopyson.file.open_(c, file_name, dirname=dir_name, generic=None, display=True, activate=True, new_window=None, regen_force=None)
    except creopyson.exceptions.ErrorJsonDecode as err2:
        print('Creoson error 2:', err2)
    except creopyson.exceptions.MissingKey as err3:
        print('Creoson error 3:', err3)
    except creopyson.exceptions.Error as err1:
        print('Creoson error 1:', err1)
```

Zepmanbc, 2022-03-17:
Hello @stekicar 
you need to catch `RuntimeError` too
```
    file_name = 'test'
    dir_name = 'test'
    try:
        creopyson.file.open_(c, file_name, dirname=dir_name, generic=None, display=True, activate=True, new_window=None, regen_force=None)
    except creopyson.exceptions.ErrorJsonDecode as err2:
        print('Creoson error 2:', err2)
    except creopyson.exceptions.MissingKey as err3:
        print('Creoson error 3:', err3)
    except creopyson.exceptions.Error as err1:
        print('Creoson error 1:', err1)
    except RuntimeError as err4:
        print('Creoson error 4:', err4)
```

stekicar, 2022-03-17:
Thank you! That was it!

closed 2022-03-17 by Zepmanbc