Zepmanbc/creopyson #47: activate particular Simplified representation

akash9470, 2022-04-06:
* Creo version:
* Creopyson version:
* Creoson version:
* Python version:
* Operating System:

### Description

i am trying to activate particular Simplified representation in creo and export it in stp format

### What I Did

```
import creopyson
c = creopyson.Client()
c.connect()
c.creo_set_creo_version(7)

c.creo_cd("O://Saran_CD//Vericut _Block Line//03_Workout//Block OP70 3D model Creation//3D models from Saran//4 CYLINDER//RE557794 (Check by Saran)")
c.file_open("re557794_mfg.prt")

c.file_list_simp_reps_(rep="op030")
c.interface_export_file("step")
```

Zepmanbc, 2022-04-06:
Hi @akash9470 
[`file_list_simp_reps`](https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.file_list_simp_reps) only list simplified reps
you should do a mapkey to activate the simplified rep
something like this:
```
c.interface_mapkey(
    f"~ Command `ProCmdViewVisTool` ;\
    ~ Select `visual_dlg0` `RadioSelApplMgr` 1 `simplified rep`;\
    ~ Activate `visual_dlg0` `Table` 2 `{simplified_rep_name}` `name_column`;\
    ~ Activate `visual_dlg0` `EvalRepRulesBtn`;\
    ~ Activate `visual_dlg0` `CloseBtn`;"
 )
```

akash9470, 2022-04-08:
i tried using same mapkey but it is not working

c.interface_mapkey(
    "~ Command `ProCmdViewVisTool` ;\
    ~ Select `visual_dlg0` `RadioSelApplMgr` 1 `simplified rep`;\
    ~ Activate `visual_dlg0` `Table` 2 `OP020` `name_column`;\
    ~ Activate `visual_dlg0` `EvalRepRulesBtn`;\
    ~ Activate `visual_dlg0` `CloseBtn`;"
 )

akash9470, 2022-04-08:
i tried using same mapkey but it is not working

c.interface_mapkey(
    f"~ Command `ProCmdViewVisTool` ;\
    ~ Select `visual_dlg0` `RadioSelApplMgr` 1 `simplified rep`;\
    ~ Activate `visual_dlg0` `Table` 2 `op020` `name_column`;\
    ~ Activate `visual_dlg0` `EvalRepRulesBtn`;\
    ~ Activate `visual_dlg0` `CloseBtn`;"
 )

DieSwartKat, 2022-04-08:
@akash9470 

try using :

```
c.interface_mapkey(
f"""~ Command `ProCmdViewVisTool` ;
~ Select visual_dlg0 RadioSelApplMgr 1 simplified rep;
~ Arm `visual_dlg0` `Table` 2 `op020` `name_column`;
~ Select `visual_dlg0` `Table` 2 `op020` `name_column`;
~ Arm `visual_dlg0` `Table` 2 `op020` `name_column`;
~ Activate `visual_dlg0` `Table` 2 `op020` `name_column`;
~ Activate `visual_dlg0` `CloseBtn`;""")
```

akash9470, 2022-04-08:
ok...
i want run that mapkey for loop and take input from FILE_LIST_SIMP_REPES module
my code is as below.........

a='OP020'

map="~ Command `ProCmdViewVisTool` ;\
mapkey(continued) ~ Select `visual_dlg0` `RadioSelApplMgr` 1 `simplified rep`;\
mapkey(continued) ~ Arm `visual_dlg0` `Table` 2 `{ab}` `name_column`;\
mapkey(continued) ~ Select `visual_dlg0` `Table` 2 `{ab}` `name_column`;\
mapkey(continued) ~ Arm `visual_dlg0` `Table` 2 `{ab}` `name_column`;\
mapkey(continued) ~ Activate `visual_dlg0` `Table` 2 `{ab}` `name_column`;\
mapkey(continued) ~ Activate `visual_dlg0` `CloseBtn`;".format(ab=a)

c.interface_mapkey(map)
.
.
..
it is not working you have any other option

DieSwartKat, 2022-04-08:
What version of python are you using?

akash9470, 2022-04-08:
python 3.7

DieSwartKat, 2022-04-08:
okay, so it seems when you passing in ab you need to do it as lowercase, not uppercase.

DieSwartKat, 2022-04-08:
I'm not using the creopyson, but my own flavour of CREOSON... here is my code

```
import BarrowsCreosonLibrary

_creoson = BarrowsCreosonLibrary()

simps = _creoson.list_simp_reps({})['reps']

for x in simps:
    simp = x.lower()
    text = f"""~ Command `ProCmdViewVisTool` ;
~ Select visual_dlg0 RadioSelApplMgr 1 simplified rep;
~ Arm `visual_dlg0` `Table` 2 `{simp}` `name_column`;
~ Select `visual_dlg0` `Table` 2 `{simp}` `name_column`;
~ Arm `visual_dlg0` `Table` 2 `{simp}` `name_column`;
~ Activate `visual_dlg0` `Table` 2 `{simp}` `name_column`;
~ Activate `visual_dlg0` `CloseBtn`;"""
    _creoson.mapkey_function(text)
    _creoson.interface_export_file("step")

```

Zepmanbc, 2022-04-08:
@akash9470 the `f` before your string is missing

akash9470, 2022-04-13:
i am getting error while exporting step file 
"not allowed characters in layer name were changed to _" in creo