Zepmanbc/creopyson #48: Change parameters of a single component instance in assembly

Krasner, 2022-05-17:
* Creo version: 8.0.0.0
* Creopyson version: 0.7.3
* Creoson version: 2.8.0
* Python version: 3.8.5
* Operating System: Windows 10

### Description

What is the approach to change a component's parameters in an assembly?
Using `c.parameter_set(file_="part.prt")` will update parameters of all instances of that part in an assembly.
How can I target a specific instance?

### What I Did

```
c = creopyson.Client("localhost", 9056)

# create new assembly file
mapkey = "Close `main_dlg_cur` `appl_casc`;"\
                 "~ Command `ProCmdModelNew`;"\
                 "~ Select `new` `Type` 1 `Assembly`;"\
                 "~ Activate `new` `OK`;"
c.interface_mapkey(mapkey)
# get assembly file handle
fa = c.file_get_active()['file']

# import components: 2 motors
out1 = c.file_assemble("para_motor.prt", into_asm=fa) #first motor results in featureid = 44
out2 = c.file_assemble("para_motor.prt", into_asm=fa) #second motor results in featureid = 45

# change parameters
c.parameter_set('TOTAL_LENGTH', 500, file_="para_motor.prt") #this changes TOTAL_LENGTH parameter on BOTH motors!
c.file_regenerate()
```

How can I use parameter_set on only 1 of the motors? Is it possible to use the featureid which is unique for each component

Zepmanbc, 2022-05-18:
Hello,
you assemble the same part "para_motor.prt" this is normal that the both assembled parts are modified.

Krasner, 2022-05-18:
I see. In this case I will make a copy of the prt with a different name.

closed 2022-05-18 by Krasner