Zepmanbc/creopyson #49: regen_failure_handling error

Krasner, 2022-05-18:
* Creo version: 8.0.0.0
* Creopyson version: 0.7.3
* Creoson version: 2.8.0
* Python version: 3.8.5
* Operating System: Windows 10

### Description
When regenerating a file with `file_regenerate()` I get the following warning:
[image]

With a message:
"Attempted use of depreciated config option regen_failure_handling is blocked"

This blocks further operation until i click OK and sometimes causes Creo to crash.
What is causing this error and how can I avoid it?

Zepmanbc, 2022-05-19:
Hello you must say to Creoson that you are using Creo 8 : https://creopyson.readthedocs.io/en/latest/usage.html#creo-7-users
`c.creo_set_creo_version(8)` for you

Krasner, 2022-05-19:
I already do that... Does it matter when this is set?
Here are my initialization steps:
```
self.c= creopyson.Client(ip_adress=host, port=9056)
if self.c.is_creo_running()==False:
    print("creo is not running, starting Creo...")
    self.c.start_creo(os.path.join(app_dir, "nitro_proe_remote.bat"))
    # allow enough time for creo to start
    time.sleep(10)
else:
    print("creo is already running")

self.c.connect()
self.c.creo_set_creo_version(8)
self.c.creo_cd(cad_dir)
```

Krasner, 2022-05-19:
updating to creoson 2.8.1 and creopyson 0.7.4 has fixed this error.

When importing components to an assembly using `file_assemble()` I still have to manually click a confirmation in creo - file_assemble will block until i've clicked the green checkbox under _Component Placement:_

[image]

Is it possible to avoid that? I can't seem to find a creo configuration setting that would skip this step.

adama2000, 2022-05-19:
When you call file_assemble(), there is a "package_assembly" option on the
creoson function.  If you set that to True then I think it will NOT prompt
you in Creo.

The "package_assembly" option is used whenever you call the file_assemble()
function without specifying any assembly constraints.  If
"package_assembly" is False then it calls the JLink function
"redefineThroughUI()" which I believe is what causes Creo to prompt you.
And False is the default value for "package_assembly" if you don't specify
a value.

Adam Andrews

Krasner, 2022-05-20:
Awesome! That works!

closed 2022-05-20 by Krasner