Zepmanbc/creopyson #50: Gathering point coordinate data

japage17, 2022-05-19:
* Creo version: 7
* Creopyson version:
* Creoson version:
* Python version:
* Operating System: windows

### Description

Fairly new to CREO automation, but is there a way to gather point coordinate data? Just looking for a list of x, y, z coordinates for a point.

adama2000, 2022-05-19:
There isn't a way to do that right now.  It could be added as a new feature
with some work, JLink supports it.

Adam Andrews

japage17, 2022-05-19:
Is that something that could be done in python? Or would more of the Java
have to be exposed to have access through jlink?

I generally write CAD automation for CATIA V5, but have been tasked with
transferring a python script I made to CREO.

Josh

adama2000, 2022-05-19:
It would have to be added to the Java code, the Python is just a wrapper
for the Java library.  The Java source code is available on the creoson
github site, if you wanted to try making the change yourself rather than
waiting for it to be part of an official release.

Adam