Zepmanbc/creopyson #54: How to get Center of Gravity Data from Mass Property

VivekPatil007, 2022-09-13:
Hello All,

I'm using Creoson for automating the data extraction from creo using python.
I need to extract mass property especially Center of Gravity but "file_massprops" is not able to get it.

Currently, it gets,
"volume, mass, density, surface area ,ctr_grav_inertia_tensor, coord_sys_inertia, coord_sys_inertia_tensor" 

Does anyone have any idea how can I get the Center of Gravity..?

Thanks In Advance 😃

Zepmanbc, 2022-09-15:
Hi @VivekPatil007 
It seems you need the BMX module to get the COG, to create the CSYS_COG and get its coordinates...
https://community.ptc.com/t5/3D-Part-Assembly-Design/Centre-of-Gravity-CG-for-parts-assemblies-and-drawings-to-be/td-p/318753
but maybe I'm wrong, if anyone knows?

I see a hard way to get it if you do a mapkey that open the mass property window, compute the properties, copy the result to clipboard, close the window.
Then you can get the result from clipboard with something like this:
`subprocess.getoutput("powershell.exe -Command Get-Clipboard")`
and then analyse the result string
Yes, this is the hard way

VivekPatil007, 2022-09-15:
Hello @Zepmanbc 
Thank You for your reply.

Is it possible to send two commands simultaneously to Creo using the Creoson server..?

if not then how I can send the mapkey to Creo from python without using hotkeys... any idea..? 

I have tried to run  c.file_assemble( ) & c.interface_mapkey('  ') commands, but at a time only one command is executing.

Zepmanbc, 2022-09-16:
You record your mapkey, get the source and use it with [interface_mapkey](https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.interface_mapkey)
You cannot send simultaneous command to Creoson

closed 2024-11-22 by Zepmanbc