Zepmanbc/creopyson #55: How to get node number for datum planes?

yuyaU0507, 2022-09-27:
Hi there, thank you for taking your time!

I'd like to automate two solidify operations to obtain a-quarter cut models using mapkey.
When I check the recorded mapkey the datum planes are defined as node numbers such as 'node3'.
The node number somehow differ from model to model, thus I'd like to pick them by using creopyson.
I've tried the 'feature_list' command with 'paths=True, inc_unnamed=True, no_comp=True' options, but the returned letters didin't contained the node numbers.

Do you have any idea to pick them?

Thanks!

Zepmanbc, 2022-09-27:
Hi
No you cannot deal with node numbers.
If you can name the datums you can do a mapkey with their name to select them (need to use creo's search feature with name to select).
I don't know if it's the right solution for your problem, it depends where your parts are from, what is it for, etc...

yuyaU0507, 2022-09-27:
Hi,
Okay I'll look for another way to do that.
Thank you!

closed 2022-09-27 by yuyaU0507