Zepmanbc/creopyson #56: Mapkey script to input a specific name?

yuyaU0507, 2022-09-28:
Hi, 

I'd like to run a mapkey script that contains a input for a search tool. The script descrives the specification as below:

mapkey(continued) `ASM_FRONT`;~ Activate `selspecdlg0` `EvaluateBtn`;

Here, the specified name "ASM_FRONT" comes before the tilde of the script. Creopyson documentation sais "the tilda must occur immediately after the semicolon at the end of the previous line", thus the specification may not be included the script.

Could we include the specification in the mapkey script for creopyson?

I'm sorry for asking several questions in a few days,
Thanks!

adama2000, 2022-09-28:
Out of curiosity, have you tried running with that script to see if it
works?

Adam Andrews

yuyaU0507, 2022-09-28:
Hello Adam,

Yes, I tried. It worked correctly except name specification.
The script contains commands to select a datum plane (using a search tool with a specified name) and perform solidify function for the plane. We had a new solidify feature but the CAD model had no change because the datum plane selection didn't work well.

Zepmanbc, 2022-09-28:
Sorry but I don't understand the question

closed 2022-10-12 by yuyaU0507