Zepmanbc/creopyson #57: creopyson.layer*() functions don't work with .drw files

tmorgan497, 2022-09-28:
* Creo version: 8.0.5
* Creopyson version: 0.7.4
* Creoson version: 2.8.1
* Python version: 3.10.5
* Operating System: Windows 11 Pro

### Description
 
Running `creopyson.layer_list()` returns nothing when the active part is a .drw file. Do the layer functions work with .drw files? 

### What I Did

```
import creopyson
creo = creopyson.Client()

# (intermediate steps)
# start creoson server
# start Creo
# open .drw file

creo.connect()
creo.layer_list()

# returns empty list, even though several layers exist in the active .drw file

creo.disconnect()
```

Zepmanbc, 2022-09-28:
Hi @tmorgan497 
Maybe I'm wrong but for me the layers belongs to the part/assembly, not the drawing
that's why the function does not return anything

tmorgan497, 2022-09-28:
> Hi @tmorgan497 Maybe I'm wrong but for me the layers belongs to the part/assembly, not the drawing that's why the function does not return anything

Hmm. Maybe it's a Creo thing, but when when I hide a layer in the part or assembly and save status, it still shows up in the drawing until I hide the layer within the drawing. I'll have to keep looking around the Creo config options, but the only related thing I've found so far is `def_shown_layers_in_view`.  Well regardless, I've been able to workaround it by integrating a mapkey into my python script to hide the layer...  

Thanks for replying so quickly!

closed 2022-09-28 by tmorgan497