Zepmanbc/creopyson #58: creopyson.bom.get_paths question

stekicar, 2022-11-07:
* Creo version: 5.0.6
* Creopyson version: 0.7.3
* Creoson version: 2.8.0
* Python version: 3.10
* Operating System: Windows 10

### Description

Extract paths for each components. I am not sure if path is components location in folder or something else.

### What I Did

```
comp_list = creopyson.bom.get_paths(srv, file_=None, paths=True, ...
```
This returns:
COMP_LIST: {'has_simprep': False, 'children': {'children': [{'children': [{'children': [{'file': 'FLAT.PRT', 'seq_path': 'root.1.1.1', 'path': [42, 42, 43]}, {'file': 'FLAT.PRT', 'seq_path': 'root.1.1.2', 'path': [42, 42, 50]}, {'file': 'FLAT.PRT', 'seq_path': 'root.1.1.3', 'path': [42, 42, 51]}, ...

I see PATH with some numbers, but I do not understand what these numbers represent. I was expecting something like a path to folder component resides.

Thank you!

Zepmanbc, 2022-11-07:
Hello

This is  hierarchy of components within an assembly
```root 1
    1st asm 1.1
        1st part of 1st asm 1.1.1
        2nd part of 1st asm 1.1.2
    2nd asm 1.2
        1st part of 2nd asm 1.2.1
etc...
```
I don't really know how to use these informations

stekicar, 2022-11-07:
Thank you for reply. I was talking about 'path': [42, 42, 51] variable. What these numbers represent?

Zepmanbc, 2022-11-07:
I don't know sorry, is there a link with the id in tree?
maybe Creoson guys knows? @adama2000 @davidhbigelow

adama2000, 2023-02-15:
The path is what's known as a Component Path in Creo.  The numbers represent the Feature IDs of the components in the hierarchy from the top of the assembly down to the given component.  You can view the Feature IDs of your components in Creo by editing the Tree Columns to add a Feat ID column.

stekicar, 2023-02-15:
Got it. Thank you! you can close this issue now.

closed 2023-02-16 by Zepmanbc