Zepmanbc/creopyson #59: freeze "file_display"

yuyaU0507, 2022-11-16:
* Creo version: 4.0
* Creopyson version: 0.7.4
* Creoson version: 2.8.1
* Python version: 3.11
* Operating System: Windows 10

Hi there,
I tried to activate a single part in an assembly using the "file_display" command as below and then python froze.
I would like to know two points:
1) Is it possible to activate a part in an assembly with the command?
2) Is the command correct?

Thank you.

---- commands ----

import creopyson
c=creopyson.Client()
c.connect()

c.file_display(file_="TEST_PART.PRT")

Zepmanbc, 2022-11-16:
Hi
Maybe I'm wrong, I never used this function, but I think your part must be in memory before you display it.
So you must use file_open before file_display.
With file_open you can display directly, that's why I never used file_display

yuyaU0507, 2022-11-16:
Hi
Thank you for sharing the mothod!

With the command, only the selected part is displayed.
I would like to activate a part with the assembly desplayed.
Is it possible to do that with creopyson?

Zepmanbc, 2022-11-17:
There is no function for this but you can do a mapkey to search the part throw its name or ID and then activate it
to run the mapkey : https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.interface_mapkey

closed 2022-11-17 by yuyaU0507