Zepmanbc/creopyson #60: Export family table as a text file to specified directory

yuyaU0507, 2022-11-21:
* Creo version: 4.0
* Creopyson version: 0.7.4
* Creoson version: 2.8.1
* Python version: 3.11
* Operating System: Win 10

Hi there,

I would like to save a family table in text format to the specified directory.
The mapkey below correctly works when I use it with Creo, but somehow does not work correctly with creopyson ("interface_mapkey" command)
Especially, the text file is saved in a different directory from the specified one. I have no idea why the file was saved there.

There is a question:
1) Is it possible to run the mapkey script with "interface_mapkey" command?

Thank you for taking your time!

-------- mapkey script ------------

mapkey f @MAPKEY_LABELExp_FamilyTable_txt;\
mapkey(continued) ~ Activate `main_dlg_cur` `page_Tools_control_btn` 1;\
mapkey(continued) ~ Command `ProCmdMmFamTab` ;~ Select `ftb_edit_table` `MenuBar` 1 `File`;\
mapkey(continued) ~ Select `ftb_edit_table` `Write`;~ Close `ftb_edit_table` `MenuBar`;\
mapkey(continued) ~ Activate `ftb_edit_table` `WriteAsText`;\
mapkey(continued) ~ Trail `UI Desktop` `UI Desktop` `DLG_PREVIEW_POST` `file_open`;\
mapkey(continued) ~ Input `file_open` `Inputname` \
mapkey(continued) `C:\\SPECIFIED DIRECTORY`;\
mapkey(continued) ~ Update `file_open` `Inputname` \
mapkey(continued) `C:\\SPECIFIED DIRECTORY`;\
mapkey(continued) ~ Activate `file_open` `Inputname`;\
mapkey(continued) ~ Select `file_open` `Ph_list.Filelist` 1 `SPECIFIED NAME.ptd`;\
mapkey(continued) ~ Command `ProFileSelPushOpen@context_dlg_open_cmd` ;\
mapkey(continued) ~ Activate `UI Message Dialog` `ok`;~ Activate `ftb_edit_table` `Cancel`;

Zepmanbc, 2022-11-21:
Hello,
I cannot test on Creo at the moment but for a mapkey  you must remove `mapkey(continued)` this is maybe the problem

yuyaU0507, 2022-11-21:
Hello,
I've removed the phrase mapkey(continued) and made them into a long script actually.
However, a text file was not saved to the specified derectory shown in the script.

Zepmanbc, 2022-11-26:
Hello,
try to debug your mapkey line by line
(sorry for the delay, it's holidays)

yuyaU0507, 2022-11-28:
Hi,
I tried a mapkey below as debug and the file was somehow exported to a different directory as a different name.

~ Activate `main_dlg_cur` `page_Tools_control_btn` 1;\~ Command `ProCmdMmFamTab` ;\~ Select `ftb_edit_table` `MenuBar` 1 `File`;\~ Select `ftb_edit_table` `Write`;\~ Close `ftb_edit_table` `MenuBar`;\~ Activate `ftb_edit_table` `WriteAsText`;\~ Trail `UI Desktop` `UI Desktop` `DLG_PREVIEW_POST` `file_open`;\~ Input `file_open` `Inputname` `C:\\creopyson_test\\test\\test.ptd`;

Could you test this mapkey on your environment?

Zepmanbc, 2022-11-30:
Sorry I'm on holidays abroad and don't have my mouse so using Creo is a bit tricky, I'll try next week when I'll be back

Zepmanbc, 2022-12-03:
Tried to do what you want and it does like you said.
I think the solution si to change working directory on the go.
You change working directory (target folder where you want to save your file), execute the table save end re-change your target directory to the previous path.
so you don't have to deal with path during the table export.

yuyaU0507, 2022-12-05:
Thank you for your check.
Do you mean I should change working directory to where I like to save file, with "creo_cd" command?
The mapkey script did not work as I thought even I changed working direcroty.

Zepmanbc, 2022-12-05:
yes, use [creo_cd](https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.creo_cd)
you can save in a variable your current working directory with [creo_pwd](https://creopyson.readthedocs.io/en/latest/creopyson.html#creopyson.connection.Client.creo_pwd)

yuyaU0507, 2022-12-05:
Tried to save a file with a changed working directory, but unfortunately, the file was not saved to the specified directory nor the changed working directory. The file was saved in the folder where the part was originally saved.

Zepmanbc, 2022-12-06:
Well, I don't know, move your file with Python

Zepmanbc, 2022-12-07:
You can contribute with a PR ;)

yuyaU0507, 2022-12-08:
Well, what shold I do?

closed 2022-12-21 by yuyaU0507