Zepmanbc/creopyson #62: Adding location to a note in a drawing

dummifiedme, 2022-12-08:
* Creo version: 4.0
* Creopyson version: 0.7.4
* Creoson version: 2.8.1
* Python version: 3.8.5
* Operating System: Windows 10

### Description
Trying to add a note at a particular location. 

But I am unable to add "location" to the note using `creoson.note_set()`. 
How to add the location? It is explicitly mentioned in the documentation, that we can add location but the argument is not mentioned!

[image]

### What I Did

I tried to add location with "point", "location" and "position" arguments. Nothing worked.

```
loc = creopyson.objects.jlpoint(190,0,200)
notee = c.note_set(name = "somethingg", point = loc, value = "Something")
```

What should I be doing?

Zepmanbc, 2022-12-09:
Sorry, never tried this, you should ask on [Creoson's page](https://github.com/SimplifiedLogic/creoson/issues) and if it's a creopyson problem i'll check it

dummifiedme, 2022-12-10:
> Sorry, never tried this, you should ask on [Creoson's page](https://github.com/SimplifiedLogic/creoson/issues) and if it's a creopyson problem i'll check it

Yes, that was what I was suggesting that creopyson doesn't have one argument for location for this function while creoson has it. If you see the second table, that is from the creoson documentation.

Also, I tried sending creoson commands using the creopyson vanilla command, and it did work fine with the added location argument.

Zepmanbc, 2022-12-10:
thanks, read your issue too fast
correction is in [0.7.5](https://github.com/Zepmanbc/creopyson/releases/tag/v0.7.5) release

closed 2022-12-10 by Zepmanbc