Zepmanbc/creopyson #63: Adding a Note in a drawing throws error in case there is a space

dummifiedme, 2022-12-20:
Creo version: 4.0
Creopyson version:
Creoson version:2.8.1
Python version: 0.7.4
Operating System: Windows 10

### Description

Describe what you were trying to get done.
I was trying to add a note into a drawing sheet and I noticed that if the string contains " " (spaces), the library will throw error.
[image]

From what I feel, the source code is trying to create a "NAME" of the note I am adding by appending the note itself to the name. There should be some sort of preprocessing while doing so. eg. the note text could be converted to non-spaced string by replacing the spaces with underscore.

Can you please fix it quickly? 
Because adding a note to drawing sheet should have capability to add any string without the limitation of keeping it free of spaces or symbols.

Zepmanbc, 2022-12-20:
are you sure the problem is in creopyson?
there is no function `create_note` and nowhere `namex` in the code

dummifiedme, 2022-12-22:
My bad.

closed 2022-12-22 by dummifiedme