Zepmanbc/creopyson #71: Export specific parameters from an open assmebly

johnmccall7, 2023-04-27:
* Creo version:6.0.5.1
* Creopyson version:0.7.6
* Creoson version:2.8.2
* Python version:3.11.3
* Operating System:windows 64

### EXPORT FULL BOM

I want to have a assembly open and be able to export a full assembly and parts(children) using specific parameters into an excel sheet.  I have done this in steps as first I used param_result = c.parameter_list(name=param_names, encoded=False) command and i was able to achieve the export of a single part or a the assembly params into an excel sheet. However the assembly children could be exported. I then used creopyson.bom.get_paths coomand and it was able to export the children part number(filenames) but the params doesnt seem to be exported using this command ether.

### What I Did( listed above)

Is there a a command to export a full BOM from an assembly ? I'm sure I have likely missed this.

Zepmanbc, 2023-04-28:
Hi,

Never tried to do this but i don't think there is a "magic" way to get all, you do it right with your steps and get params for each part.
Did you have a look at this? https://www.simplifiedlogic.com/nitro-bom
Never tried but this is made by Creoson's team

DieSwartKat, 2023-04-29:
I've build a tool to help with this. It is not using creoson though, but with the help of a client that runs some mapkeys it extract the data from the assembly and upload it to a website called creobuddy.  Creobuddy then allow you to filter out any items you do not want to see based on a few simple rules. You can then download your data to CSV.

It's not perfect and still very rough around the edges. Plenty functionality still missing. I've been trying to get people to use it to get feedback so that I can priorities functionality. If you are interested you can view it on [creobuddy](https://creobuddy.onrender.com/)

davidhbigelow, 2023-04-29:
@DieSwartKat 

WOW - VERY COOL!  

Love the UX for your interactions via Web Page!  Super clean and intuitive!  (Well done!)

Would be curious as to the interface you are using to transact with Creo - all Mapkeys?  Seems like some of this is more like a Pro/TOOLKIT or API thing.

It appears that you are pushing data to the cloud... That will be a significant barrier for many companies.  Recommend you have it run locally for all data transactions and exports.  That would be more secure (assuming my view of this is correct).

Open Sourced?

DieSwartKat, 2023-04-29:
@davidhbigelow 

I'm making use of the mapkeys only, not API or Pro/TOOLKIT. Which is why I need people to test to it. I've tested it on CREO 3 and still need to test on other platforms.
I'm aware the uploading data is a security risk that many companies are not willing to take. The challenge is to get a locally run program with a simple interface as well as securing the source code since it's not opensource.
I've actually taken the tool offline, but reactivated it when I saw this question.
Thanks for the feedback.

davidhbigelow, 2023-04-29:
@DieSwartKat 

I think what you have is very cool.   Local execution is part of why CREOSON was implemented the way it was.   You can touch a Creo session locally or over the network (permissions obviously) and provide a remote interface to Creo data.

You could easily leverage the CREOSON server with your own local Web Page to do the same thing.   Since you are using only Mapkeys - you could send/received them through the local web-service and effectively get the same result.

Recommend you check into how we do this.   You can use CREOSON directly - and just add your calls to it.  The Sandbox is a good place to test/debug, you can put your own HTML in the Web Server directory and run it from within CREOSON -- you just need to hook into the CREOSON Mapkey call to send the request and catch and display the responses in your UI.

http://www.creoson.com/functions.html

After you get the session connected - you can run the following command to transact the data:

interface : mapkey

You might consider the JavaScript interface since you are already in a Web Page - but Python can do the same thing if that is more comfortable.

The only thing I had a concern with is that the local mapkeys being executed from a cloud env and pulling data back for display and execution.   Corporate people would frown on that data being pulled out of their networks to an unknown endpoint.  (just saying).  Not sure that is how this is working -- but that is what caught my attention.   If that is the direction you are thinking - then make it very clear that is what is happening -- SSL all the transactions for additional security.

Pretty cool stuff!

johnmccall7, 2023-04-29:
>

johnmccall7, 2023-04-29:
Thanks for replying everyone. I have a basic knowledge of "coding" and been an design engineering of heavy equipment for over 25 years . I have used Solidworks for 15years and used a software called batchprocess to help export the BOMs.I worked with the programmer to quash all the bugs..  I have been using Creo for 10years and never had a similar software so took it upon myself to sort the design office issue. I appreciate every company different and have different attributes etc. However  my aim was to export the boms into 7  separate BOM for each of the Manufacturing Processes.. Unfortunately Creo son /creopython do not have / allow this function available. As mention above they have a software nitro. Doesn't represent what I what to achieve...  
Which was have one mapkey BOM export/ find all PDF/DXF for parts in a live model. 

Personally I have gave up trying to achieve this with creson/ creopython.. as @davidhbigelow  mention there no magic command.. I just used python script .. I just export the full BOM( 5-10k part assemblies )from drawing table to csv and from this I can create my 7 BOM and can search the network drives for all the pdf/ dxf for the latest version and save back to the 7 different folders .. it not what I wanted at the start  working directly within Creo.. but I hit a wall and did it this way as I have all the details from the model then I can run everything....

@DieSwartKat  thank you  for your suggestions.. I like what you have done.I am 100 percent sure any company wants their data on their server. Speaking from experience being in those meetings. You have to make that the software  can work for all companies. You seem to have different filter properties.. if there was a way to store reports  so you can just select them straight away.. from those reports you can batch print them or send them to a folder. Have ways to search the model for missing drawings or dxfs. Or search the model for missing properties etc. These all help. 

These are all feature you may have thought of or have. I just know they can help save time and remove errors always help.  Thanks guys for making suggestions it is much appreciated you taking the time. ..

DieSwartKat, 2023-04-29:
@johnmccall7 out of curiosity, how much time will it take to export a 5 to 10K part assembly using batchprocess in SolidWorks?

johnmccall7, 2023-04-29:
> @johnmccall7 out of curiosity, how much time will it take to export a 5 to 10K part assembly using batchprocess in SolidWorks?

Hi, when I used Solidworks I was making machines with only 2000-4000 but It was instant.. basicly how it worked you have defined report for example Saw in a html File and user could change the the report format in that and when you selected that pre defined report it would create it instantly.. 

It also could batch print etc. I'm sure there is YouTube videos . There a free trial to use it from memory. It my angelsix.         https://store.angelsix.com/BatchProcess-p141128027

johnmccall7, 2023-04-29:
@DieSwartKat  out of my curiosity.. did you create your own code to transverse the assembly into the children obtain all the parameters? What code language did you use?

closed 2024-11-22 by Zepmanbc