Zepmanbc/creopyson #72: Error in drawing_create

jordifas, 2023-05-18:
* Creo version: 6.0.6
* Creopyson version: 0.7.6
* Creoson version: 2.8.2
* Python version: 3.11
* Operating System: windows 10

### Description

I try to create a new drawing from a model. Everytime I get this error:

### What I Did
I ran:
import creopyson

c = creopyson.Client()
c.connect()

filename = creopyson.file.get_active(c)["file"][:-4]
model = creopyson.file.get_active(c)["file"]

c.drawing_create("Template", model= model, drawing= filename + ".drw" , activate= True)

I got this error:

  File "C:\Program Files\Python311\Lib\site-packages\creopyson\connection.py", line 75, in _creoson_post
    raise RuntimeError(error_msg)
RuntimeError: A Pro/TOOLKIT error has occurred: General Error.  Check your model/drawing for accuracy.
```

Zepmanbc, 2023-05-20:
Hello
Did you try with "[vanilla creoson](https://creopyson.readthedocs.io/en/latest/usage.html#vanilla-creoson-usage)"?
I don't think the problem is in creopyson regarding this error message

DieSwartKat, 2023-05-20:
It's worth trying to increasing the accuracy of the CREO model. Sometimes CREO has a problem calculating some geometry and you get a similar error.

bperryrm, 2024-02-05:
I'm able to get this to work using:

filename = creopyson.file.get_active(c)["file"][:-4]
model = creopyson.file.get_active(c)["file"]
c.drawing_create("template_name.drw", model, "test.drw", 1, True)

closed 2024-02-09 by jordifas