Zepmanbc/creopyson #74: Suppressing parts in a subassembly with using feature_suppress

yuyaU0507, 2023-06-01:
* Creo version: 4.0
* Creopyson version: 0.7.6
* Creoson version:2.8.1
* Python version: 3.11
* Operating System: Win 10

Hi there,
I would like to suppress a part included in a subassembly and tried feature_suppress command as below:

c.feature_suppress(name='PARTNAME.prt')

This command did not work, even it works when I change the name to subassembly name or part name that is not included in the subasesmbly.
Could you tell me if I would have a way to do that? (Ex. add options. or activate the subassembly first..etc.)

Thanks!

Zepmanbc, 2023-06-01:
Hi yuyaU0507

Never tried this but it seems that you cannot suppress something in a subassembly because it is not a part of the main tree, the «feature» functions deal with the tree.

I think the right proccess should be:
1. Open the subassembly (display is not needed)
2. Suppress the file (it is now in the current main tree)
3. Close the subassembly (and reactivate your main window if needed)

Hope it helps

yuyaU0507, 2023-06-02:
The process works.

Thanks a lot!

closed 2023-06-02 by yuyaU0507