Zepmanbc/creopyson #75: How to export tolerance (Geometrical tolerance & Measurement tolerance) and the related links

Thai-NX, 2023-06-08:
* Creo version: Creo 9
* Creopyson version: Version:0.7.6
* Creoson version: Version 2.8.2
* Python version: Version:3.11.3
* Operating System:  windows 10 x64

### Description

Hello everyone, i'm new to Creopyson.
I have a task want to hearing your advice:
I want to export all tolerance in DWG to excel sheets. 
Want to get their information, and the link related if it had.
[image]
Because i want to check the tolerance easier way, don't need to look at drawing, 
so i need to export the tolerance information, each by each, and what link to it.
By checking in excel sheet, i can easily check which tolerance and things related to it
If you all have any ideas for this, I would greatly appreciate it.
Thank you so much.

### What I Did

```
Is there a a way to export tolerance from DWG to excel sheet? i'm new to Creopyson.
```

Zepmanbc, 2023-06-09:
Hi,
Never have to do something like this, sorry.
Maybe someone else will help you

Thai-NX, 2023-06-09:
Thanks  for your comment.
Are there any one else can help me with this, please.
Or email me privately: xuanthai89@gmail.com
Thanks.

closed 2024-11-22 by Zepmanbc