Zepmanbc/creopyson #76: Request add Drawing sheet information into "Note.py" in "List" function.

Nam-Hoang68, 2023-06-12:
Creo version: Creo 9
Creopyson version: Version:0.7.6
Creoson version: Version 2.8.2
Python version: Version:3.11.3
Operating System: windows 10 x64

### Description
I am using "Note.py" of CREOPYSON to extract Geometric tolerances from Multi-sheet CREO Drawing.
The results when using "Note.list" function did not return the Drawing sheet ID information that has Geometric tolerances.
For reference when using "Drawing.List_symbol" function, the results included Drawing sheet ID information.

### What I Did
Could you please help me modify the "Note.list" function, that can return the Drawing sheet ID.

Thai-NX, 2023-06-12:
For exmple: 
here is the returns of Note.List: lists Notes
{'location': {'x': 493.9712844537856, 'z': -2.5898961071027332e-15, 'y': 133.9202192358037}, 'name': 'gp54', 'value': '&G54:6', 'encoded': False}   
[image]
[image]

Could you add the sheet number information in Note.List: lists Notes

Zepmanbc, 2023-06-13:
creopyson is just an overlay of Creoson, it returns what Creoson returns.
You can ask Creoson's team but I think it will be easier to create your own functions with differents queries to combine all the data you need.

Thai-NX, 2023-06-14:
> creopyson is just an overlay of Creoson, it returns what Creoson returns. You can ask Creoson's team but I think it will be easier to create your own functions with differents queries to combine all the data you need.

Thank for your reply. 
Can you teach how to create the simple function, i'm jusst an CAD enineering, not much experiment in python or coding somthing.
Or you can mail me: xuanthai89@gmail.com
Thanks.

closed 2024-11-22 by Zepmanbc