Zepmanbc/creopyson #78: Could I replace generic instance by using "familytable_replace"?

yuyaU0507, 2023-07-19:
* Creo version: 4.0
* Creopyson version: 0.7.6
* Creoson version: 
* Python version: 3.11
* Operating System: Win10

### Description

Hi there,

I'm tring to write a code to switch familytable instances using "familytable_replace" command.
In this procedure, I'd like to replace an instance to "generic" one but the program returns " No instance 'generic' in model XXX.PRT".
Then I checked instance names by using "familytable_list" command and found that "generic" instance is not contained in the list.

Is there any method to change an instance to the generic?

Thank you for your help!

Zepmanbc, 2023-07-19:
Hello
familytable_replace is for replacing with an instance of the same family table, I think the generic is not considered as an instance, you should create an instance with all variable values as *

yuyaU0507, 2023-07-21:
Okay,
I will create an instance with same variable values as the generic.
Thank you!

closed 2023-07-21 by yuyaU0507