Zepmanbc/creopyson #79: How to do "save as" with creopyson?

yuyaU0507, 2023-10-18:
* Creo version: 4.0
* Creopyson version: 0.7.6
* Creoson version: 2.8.1
* Python version: 3.11
* Operating System: Win10

### Description

I would like to save a CAD file with different name. I found "file_save" command but the function seems to overwrite the file.
Do you know any other functions that save files with different names?
Should I use mapkey instead?

Thanks!

Zepmanbc, 2023-10-18:
Hi,
Yes you should use mapkey
Save as should be complex for ASM, this should be the reason there is no build in save as function

adama2000, 2023-10-18:
I believe that file:backup might work.  Not at my computer right now or I'd
check.

yuyaU0507, 2023-10-26:
Hi,
I tried "file_backup" to save a file to other directory then it works.
I'll rename the file with commands. 
Thanks

closed 2023-10-26 by yuyaU0507