Zepmanbc/creopyson #83: "feature_suppress" for parts in subassembly

yuyaU0507, 2024-01-26:
* Creo version: 4.0
* Creopyson version: 0.7.6
* Creoson version: 2.8.1
* Python version: 3.11
* Operating System: Win10x64

### Description

Is it possible to suppress features (parts) included in a subassembly by using "feature_suppress"?

### What I Did

Model configuration:

model.asm
　┣subasm.asm
　┃　┣subasmmodel.prt
　┃　┗subasmmodel2.prt
　┣model2.prt
　┗model3.prt

I tried feature_suppress(name="subasmmodel.prt") to suppress the part remained active.
Suppressing other part (for model2.prt or model3.prt) successfully worked.

Should I add variables to the command?

Thanks!

Zepmanbc, 2024-01-26:
Hi, the feature does not belong to the opened model so you cannot supress it like in GUI
You must open the part/asm  in which the feature/part is referenced

You can open your subasm without show it, suppress what you want, regenerate, close. I'ts quick so you can think it would have been done it without open subasm.asm

yuyaU0507, 2024-01-26:
Thank you for your suggestion!
I tried commands below but models were not seemed to be suppressed.

file_open(file_="subasm.asm", display=False)
feature_suppress(name="subasmmodel.prt")
file_regenetrate(file_="subasm.asm")
file_close_window(file_="subasm.asm")

Would you know they are correct?

Zepmanbc, 2024-01-27:
you probably need to regenerate model.asm too

Zepmanbc, 2024-01-27:
it's been a long time I did not look at creopyson, sorry
if that does not work, you should have to do this with mapkeys
 * select the part you want (but you will still need to open subasm and look after subasmmodel with the mapkey)
 * then a mapkey that suppress the selected item in tree

yuyaU0507, 2024-03-21:
Finally, I found these commands below works:

file_open(file_="subasm.asm", new_window=True, display=True)  
feature_suppress(name="subasmmodel.prt")
file_close_window(file_="subasm.asm")
file_regenetrate(file_="subasm.asm")
file_open(file_="model.asm", new_window=False, display=True)

Thank you for your advices.

closed 2024-03-21 by yuyaU0507