Zepmanbc/creopyson #89: Creo4 ，in not network , execute file_regenerate() or file_regenerate(file_name) Stuck in regenerating progress

maipj, 2024-09-05:
* Creo version: 4.0
* Creopyson version: 0.7.7
* Creoson version: 2.8.1
* Python version: 3.11
* Operating System: Win10x64

### Description

use file_regenerate  and Stuck in regenerating progress ，The program runs without a network connection

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
creo_client.dimension_set(name=item[0], value=val, file_="work_file.prt")
creo_client.file_regenerate("work_file.prt")
```

When I have a network, executing file_degenerate() will succeed, but when the network is disconnected, there will be a situation where the progress is regenerated and stuck

I checked the network and modified the virtual network card to add creo4 to the virtual network card, but the above situation still occurs, The program runs without a network connection

What do I need to do? How do I implement file_degenerate() when the network is disconnected

Zepmanbc, 2024-09-07:
I think this is more a question for Creoson
But I remember that I mainly use a mapkey to regenerate more than the file_regenerate

adama2000, 2024-09-08:
Nothing about creoson tries to connect to the network, so I don't know why
that would make a difference for it.  Something about Creo itself, maybe?

One suggestion is that there might be errors popping up in the creoson
console.

Try stopping the creoson server, then opening a command prompt and running
it manually (creoson_run.bat).  If an error occurs during regenerate, it
might pop up there.

Another question is, in the Creo window do any errors appear, like in the
status bar at the bottom?  Or does the whole Creo window lock up when this
happens?

maipj, 2024-09-09:
> Nothing about creoson tries to connect to the network, so I don't know why that would make a difference for it. Something about Creo itself, maybe? One suggestion is that there might be errors [...]

yes , this is some picture
[image]
[image]
[image]

Stuck on this interface P1
and the P2 say the regeneration is successfully
I don't know!! is crazy!!

maipj, 2024-09-09:
yes I use cmd to run creoson_run.bat , and other creopyson is ok, buy this regeneration is Stuck on that interface!! I must use Window Task Manager To stop and remove Creo4

closed 2024-11-22 by Zepmanbc