Zepmanbc/creopyson #91: Bill of Materials export text file

Quatannens, 2024-12-02:
* Creo version: 10.0.5.0
* Creopyson version: 0.7.7
* Creoson version: 2.9.0
* Python version: 3.13.0
* Operating System: Windows 10 Pro

### Description

I would like to create a .bom file from the active model in Creo. I already have a config option that sets the path to a certain format file. But I don't see how I can use the functionality of Creo to export the bom to a file using Creopyson.
The desired .bom file has a certain format that is necessary for other actions later on (fixed width and delimiters and necessary parameters).

### What I Did

I tried using get_bom_paths, but this gives a lot of information I don't need and it misses the parameters I do need. I tried exctracting parameters using parameter_list, but this also gives a lot of data to process. I feel there has to be an easier way. A command that will activate 'Bill of Materials' in Creo.

adama2000, 2024-12-04:
You will need to process the output from the cropyson calls before you can
write to your output file, to get it into the specific format you need.

get_bom_paths outputs a tree structure of your assembly.  What you do with
that depends on what you need.  For example, if you just want a flat list
of parts with quantities, your code will need to walk down the tree
structure and accumulate that list in memory; then at the end of walking
the tree, you'd write the accumulated list out to the .bom file in the
appropriate format.

For parameters, get_bom_paths doesn't output any parameters.  For that you
need to use the parameter_list command.

   - Since parameters are the same for every instance of a component, you
   only need to call the function once per component.  Therefore you should
   get the list of unique components first (as explained above) and loop
   through them to make the parameter_list call so you don't make more calls
   than you need to.
   - If you want to get specific parameters, you can use the "names"
   argument to parameter_list to specify the parameters you want returned.  if
   you only want one parameter, you can just specify that with the "name"
   argument.  If you don't specify either the "name" or "names" arguments,
   parameter_list will return _all_ the parameters on the component.

If you do this to get the list of components and their parameters, you
should be able to write code to generate the file in the format you want.

Does that help?  If not, let us know.

Adam Andrews

Quatannens, 2024-12-04:
I am able to get a structure of the Bill of Materials, just like what Creo exports. It's just that for big assemblies this takes a very long time compared to hitting the BoM button in Creo. I get the list with get_bom_paths, but I have to do a lot of adjustments before I can use the list.
- Recursive function to remove all the nested combination of lists and dictionaries
- Loop over the flat list and add all the parts and subassemblies to their corresponding assembly. While doing this I also have to search for parts and assemblies that occur more than once. I get a list with for example: Assembly1 contains Part1, Part1, Part1. I add up the same parts and assemblies to shorten the list.
- Loop over the list and call parameters.
This already takes up a minute for a small assembly (250 parts/assemblies) and because there are some nested loops, the time will go up exponentially. (The minute is without taking all parts and assemblies in memory)
If you have a bom format and export to a .bom file with Creo it takes less than a second.

So yes, I know it's possible, but it takes way longer and was hoping I could just use the Creo functionality.

adama2000, 2024-12-10:
Yes, sorry, it's written that way because we wanted to know the hierarchy
of assemblies and parts, not just get a flat list of parts like Creo
provides.

You could just list open files, but that would not give you quantities.

Adam

Quatannens, 2024-12-18:
Ok, thanks for the clarification.

closed 2024-12-18 by Quatannens