Zepmanbc/creopyson #92: How to obtain configurations of a feature such as chamfer size?

yuyaU0507, 2024-12-03:
* Creo version: 4.0M060
* Creopyson version: 0.7.7
* Creoson version: 0.2.8
* Python version: 3.11.4
* Operating System: Win10

### Description

I would like to know the configurations of each feature. For example, it would be a great help if it could take hole diameter or chamfer size by creopyson.
I tried "feature_list() " to obtain a list of the features and "feature_set_param() " to see what parameters contained the feature. However, it returned the parameter of the feature, not the configurations.

adama2000, 2024-12-04:
feature_set_param() would _set_ a feature parameter, you probably mean
feature_list_params.

"parameter_list()" would give you the parameters on a component, but how
would you tell whether a parameter contained the feature?  Would it be
based on the name of the parameter, or the value of it?  Either way, you
should be able to get that with parameter_list().

Adam Andrews

yuyaU0507, 2024-12-04:
Hi,
Yes, parameter_list() shows parameters contained in a feature.
However, I would like to obtain the length of a chamfer (e.g. pick 10 from 10x45° chamfer) by creopyson.
Is it possible?

adama2000, 2024-12-10:
We can get Parameters, and we can get Dimensions.  Is there a Dimension on
the part which defines the chamfer's 10" dimension?  You could try the
creoson "dimension_list() on the part to see what kind of results you get.

Though even if it is available as a Dimension, it may not always have the
same name on different parts.  Depends on how your models are set up.

If it's not stored in a Parameter or a Dimension, I don't know how you
would get it.

Adam Andrews

yuyaU0507, 2024-12-11:
Hi Adam,
Thank you for sharing the function. I tried it and found there are no specific definitions of the dimension actually.
Thus it isn't easy for me to identify which dimension indicates the slope size of the chamfer with the function. 
It would be better for me to code a new program that obtains the dimension value with CreoToolkit.
Thanks.

closed 2025-11-08 by Zepmanbc