Zepmanbc/creopyson #95: add ability to export pdf of a drawing file

bperryrm, 2025-05-13:
When exporting a drawing file (*.drw) as PDF, I'd like to be able to specify the "monochrome" option in addition to the "color" and "greyscale" options. The "color" and "greyscale" options are currently part of the "use_drawing_settings=True" option in the creopyson.interface_export_pdf function

davidhbigelow, 2025-05-13:
You might be able to set that in creo config print options or config
settings before you do that export.

Just a thought

bperryrm, 2025-05-14:
> You might be able to set that in creo config print options or config
> settings before you do that export.

This is actually a good idea. I was able to set config option pdf_use_pen_table and then create a pen table with black pens only. This lets me export a DWG file as monochrome color.

davidhbigelow, 2025-05-14:
Awesome, thanks!